jacksonliam/mjpg-streamer #32: CORS support

jacksonliam, 2016-08-03:
Need support for not tainting canvas when pulling image cross origin.

moppymopperson, 2018-07-14:
Has there been any progress on this? I'm working on a project where I am rendering the received images into an HTML canvas, but it's being blocked because of CORS.

jacksonliam, 2018-07-14:
There is an `Access-Control-Allow-Origin: *` header in the HTTP responses (https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_http/httpd.c#L437) but its not working and I'm not sure why!

I don't have time to debug it right now, do you happen to know why its not working?

moppymopperson, 2018-07-14:
No, unfortunately it's pretty far outside my field of expertise. I'll see if I can figure it out, but don't cross your fingers.

moppymopperson, 2018-07-14:
Here's somebody else having a similar issue, though they are probably not using mjpg-streamer. Could it be a browser bug? That seems extremely unlikely.
https://stackoverflow.com/questions/39526467/using-mjpeg-stream-as-webgl-texture

moppymopperson, 2018-07-14:
It looks like you may need a second header. This reference is setting two headers, the second for allowed content types. I'll try to find time to test it out later this week if you don't beat me to it.

https://enable-cors.org/server_expressjs.html
```
res.header("Access-Control-Allow-Origin", "*");
res.header("Access-Control-Allow-Headers", "Origin, X-Requested-With, Content-Type, Accept");
```

wwladikw, 2018-11-11:
Unfortunately, I am still not able to get CORS working - even not with NGINX proxy posted by @moppymopperson above.

@moppymopperson , could you tell me in more detail how you got it running?

What I am trying to achive is to get username/password filled by php formuler I am using in the PHP session. I have an javascript ajax call which looks like:

```
$(document).ready(function auth_media() {
	var media_url = "http://IP:PORT/?action=stream";
	var output = '<img width=80%  height=auto src=' + media_url + ' >';
	$.ajax({
	    url: media_url,
	    username: "XXX",
	    password: "YYY",
	    error: function () {
		alert("Authentication failed!");
	    }
	}).done(function(html) {
		$("#show_media_stream").html(output);
	});
});

```
Thanks a lot!

moppymopperson, 2018-11-12:
@wwladikw I have a working project up on GitHub that you are welcome to use as a reference if you’d like.

https://github.com/moppymopperson/senseekan/blob/dev/nginx.conf