jacksonliam/mjpg-streamer #37: can you add a imagemagicky processing plugin?

yunfan, 2016-08-15:
the origina needs is that i want to add datetime for my web camera's stream

then i think if this shouldnt be written for such special target, it should be implement as a common usage

so i remind that imagemagick has provided lots of small tools to manipulate images, and it works well, also well known to others. so it will be very useful if you wrote a imagemagick plugin for processing the stream and handle it to the output

also i think its better to adjust the flow model from "input->output"  to  "source->handle1->handle2->destination"   so people could write plugins much more easilly

jacksonliam, 2016-08-15:
It's a nice idea and a thought I've had before. But I don't think imagemagick is perfomant enough. 

I think it can be added to the raspicam plugin using the text code from raspivid, something I'd love to take pull requests for but it's not on my to do list. 

It also might be possible to sneak in some text using a library to write to the yuv buffers when using the -y flag on the input uvc plugin. I'd take pull requests for this too. 

Intermediate Plugins are not the best idea for mjpg-streamer because the buffer between plugins is JPEG. So each plugin would need to decode and re-encode the JPEG. On anything equal or less powerful than a raspberry pi simply encoding each frame in software takes up too much resources for a reasonable framerate and resolution. So the main target and most of the user base for mjpg-streamer (low resource/embedded) wouldn't even be able to use the feature. There could be some optimisation for sources which can provide uncompressed frames but some sources (some USB cams, IP cams) can only provide JPEG, also limiting who would be able to use the feature. 

So I think it would be better to add functionality to input/output plugins where it makes sense and is most optimised. 

Gstreamer (gst-launch)  provides the sort of complex source/sink design you mentioned and may be able to suit your purposes here. But because of this design there's a bit of a learning curve.

closed 2016-08-15 by jacksonliam

yunfan, 2016-08-15:
i am sorry, so currently, i cant add text event using -y flag on the input uvc plugin?

jacksonliam, 2016-08-15:
No, not currently and I don't plan to.
But if you or someone else wants to do the work I'll happily take a pull request!

virtuald, 2016-08-15:
You can do any kind of rewriting operations that you want if you use the [OpenCV input plugin](https://github.com/jacksonliam/mjpg-streamer/tree/master/mjpg-streamer-experimental/plugins/input_opencv)... you just have to take the time to write the code to do so.

yunfan, 2016-08-16:
ok i got it. its opencv puttext. sorry for boring