jacksonliam/mjpg-streamer #38: i got error while compiling the whole code

yunfan, 2016-08-16:
OS: ubuntu 16.04 x86_64

and the compiling log

```
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found version "3.5.2") 
-- Found PythonInterp: /usr/bin/python3.5 (found version "3.5.2") 
-- Found NUMPY: /usr/lib/python3/dist-packages/numpy/core/include  
-- Found components for NumPy
-- NUMPY_ROOT_DIR    = /usr/local
-- NUMPY_INCLUDES    = /usr/lib/python3/dist-packages/numpy/core/include
-- NUMPY_LIBRARIES   = 
-- NUMPY_API_VERSION = 1.11.0
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libgphoto2'
--   Found libgphoto2, version 2.5.9
-- Found GPHOTO2: /usr/lib/x86_64-linux-gnu/libgphoto2.so  
-- Looking for include file linux/videodev2.h
-- Looking for include file linux/videodev2.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found SDL: /usr/lib/x86_64-linux-gnu/libSDLmain.a;/usr/lib/x86_64-linux-gnu/libSDL.so;-lpthread (found version "1.2.15") 
-- 
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE , File input plugin
 * PLUGIN_INPUT_HTTP , HTTP input proxy plugin
 * PLUGIN_INPUT_OPENCV , OpenCV input plugin
 * PLUGIN_CVFILTER_CPP , OpenCV example filter
 * PLUGIN_CVFILTER_PY , OpenCV python filter
 * PLUGIN_INPUT_PTP2 , PTP2 input plugin
 * PLUGIN_INPUT_UVC , Video 4 Linux input plugin
 * PLUGIN_OUTPUT_FILE , File output plugin
 * PLUGIN_OUTPUT_HTTP , HTTP server output plugin
 * PLUGIN_OUTPUT_RTSP , RTSP output plugin
 * PLUGIN_OUTPUT_UDP , UDP output stream plugin
 * PLUGIN_OUTPUT_VIEWER , SDL output viewer plugin

-- The following OPTIONAL packages have been found:

 * OpenCV
 * PythonLibs
 * PythonInterp
 * Numpy
 * Gphoto2
[... 99 lines cut ...]
                                ^
/home/jyf/repo/git/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:403:5: note: in expansion of macro ‘CVOPT_SET’
     CVOPT_SET(CAP_PROP_GAIN, gain, "gain")
     ^
/home/jyf/repo/git/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:404:15: error: ‘CAP_PROP_EXPOSURE’ was not declared in this scope
     CVOPT_SET(CAP_PROP_EXPOSURE, ex, "exposure")
               ^
/home/jyf/repo/git/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:388:32: note: in definition of macro ‘CVOPT_OPT’
         if (!pctx->capture.set(prop, settings->var)) {\
                                ^
/home/jyf/repo/git/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:404:5: note: in expansion of macro ‘CVOPT_SET’
     CVOPT_SET(CAP_PROP_EXPOSURE, ex, "exposure")
     ^
plugins/input_opencv/CMakeFiles/input_opencv.dir/build.make:62: recipe for target 'plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o' failed
make[3]: *** [plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o] Error 1
make[3]: Leaving directory '/home/jyf/repo/git/mjpg-streamer/mjpg-streamer-experimental/_build'
CMakeFiles/Makefile2:237: recipe for target 'plugins/input_opencv/CMakeFiles/input_opencv.dir/all' failed
make[2]: *** [plugins/input_opencv/CMakeFiles/input_opencv.dir/all] Error 2
make[2]: Leaving directory '/home/jyf/repo/git/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:127: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/jyf/repo/git/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 2
```

jacksonliam, 2016-08-16:
Do you have opencv 3.1.0 installed?

yunfan, 2016-08-17:
nope, the verison of opencv in ubuntu1604's repo is 2.x  and since i saw that example_filter.py    has a `import cv2` , so i just install that opencv2.x from repo

jacksonliam, 2016-08-17:
Documentation specifically says it won't work with opencv 2.x
https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_opencv/README.md

Does it compile if you install 3.10?

yunfan, 2016-08-17:
@jacksonliam will install 3.10 after my back home. if i have compiled it. should i modify the exmple python filters from `import cv2` to `import cv3`  ?

virtuald, 2016-08-17:
Do you need opencv (are you trying to create a filter plugin)? If not, then you probably can just disable it via a cmake command. It's an optional dependency, and the rest of mjpg-streamer will work just fine without it.

No need to rename the imports, it's still cv2

yunfan, 2016-08-17:
please close this. i just gave it up.

jacksonliam, 2016-08-17:
@virtuald can cmake be set to only try to compile the opencv plugin if opencv > 3.X is installed?

virtuald, 2016-08-18:
I don't have time now, but I can definitely do it in the future.

jacksonliam, 2016-08-18:
Cool thanks! I'll have a look too, was just checking if you think it's possible!

closed 2016-08-18 by jacksonliam