jacksonliam/mjpg-streamer #42: -usestills option behaviour seems to have changed.

nophead, 2016-09-19:
Around May 2014 I successfully used -usestills to reduce the load on the CPU of an RPI B by a factor of about 10. It seemed to also have the side effect of reducing the field of view.

When I use it now with a more recent version shipped with OctoPi it only reduces the data by about 50% but it also makes the picture much darker and with a yellow cast and no longer changes the FOV.

[image]
[image]

Have there been any deliberate changes to the behavior? . Perhaps the smaller FOV was the reason it reduced the data so much.

jacksonliam, 2016-09-19:
Can you show me the two different commands you used? 

Usestills, takes a series of still pictures one after the other. The maximum framerate is about 8, which may be why you saw a reduction in data. There may also be some differences in the default resolution and jpeg quality settings between the two modes. 

The field of view depends on whether the full sensor is used or if a crop is taken. That depends on output resolution and aspect ratio. Both stills mode and (since the pixel binning enhancement) video mode can use the full sensor. 

IIRC, in stills mode, the white balancing and exposure is calculated for each frame, though this is abstracted in the MMAL API's and lots of changes have gone on in this area behind the scenes. The changes you're seeing here may be actually to do with having an updated raspberry pi firmware and not mjpg-streamer. 

The CPU use I see from mjpg-streamer with the raspicam is only a few percent, all of the image stuff is done in the GPU, so I'm surprised it's an issue. 

Overall unless you want resolutions above 1080p or very low framerates with manual control over exposure (e.g for low light shots) I'd be sticking to video mode (not usestills) and tweaking the resolution and framerate (and possibly jpeg quality) to achieve your desired data rate. 

See docs here https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/README.md

nophead, 2016-09-19:
The command line is just
```
./mjpg_streamer -o output_http.so -w ./www -i input_raspicam.so -fps 24 -x 640 -y 480
```

When running OctoPrint the extra load on the CPU when streaming video is about 75% and the total hits 100%. A lot of the extra CPU is in haproxy and kworker processes. mjpg_streamer is only about 12% so I think a lot of the CPU is simply shifting the large amount of data over the network.

The video lags by 35 seconds when using a WIFI adaptor, I have to use a direct link. The network traffic is 25mbps dropping to half that with usestills. On my old machine it drops to about 2.5mbps and the CPU load is only a few percent as you say.

I would love to stick with video mode but it cripples the PI B with a raspicam, works fine with a Logitech USB camera and it used to work with -usestills.

It looks like is going to be very hard to find out what has changed.

jacksonliam, 2016-09-20:
Okay IIRC 25Mb/s is the max bitrate for the mjpeg encoder. I think what's happened is I need to actually set the bitrate to 0 (rather than leaving it as the default) so that its dynamically calculated by the jpeg quality, like for stills. 

It's possible the default bitrate is now, or always was, 25Mb/s.

...and I might as well add the -b flag like raspivid while I'm at it! 

Can you test that the quality argument currently causes no change to the bit rate in video mode? What happens if its set to 0?

nophead, 2016-09-20:
Well this morning things are more confusing! I have two RPI Bs with the cameras  mounted next to each other looking at roughly the same scene. Yesterday they produced the same data rate and -usestills had the same effect of halving it and changing the exposure. This morning one behaves exactly the same as last night but the other has changed. The data rate is now only 10mbps. -usestills seems to have no effect at all. It doesn't affect the data rate and it doesn't change the exposure.

Reducing quality reduces the data rate but setting it to 0 causes the process to terminate after the "Starting Camera" message. Setting it to 1 gives a very low quality image and data rate.

Getting rid of usestills and setting quality to 10 gives a data rate of 0.64 mbps and reduces the CPU overhead to about 25% but the picture quality isn't great. It doesn't work as well as my old setup.

I don't understand why one is behaving differently and the other is the same. The lighting has changed: last night was totally fluorescent but today there is some daylight as well. 

It also seems that I get worse performance than most people. For example a friend here https://groups.google.com/d/msg/octoprint/CPF4XVTxDeg/43nR4MkNmQsJ can run at 24 fps without using much CPU.

jacksonliam, 2016-09-20:
Are they running the same rpi firmware and mjpg-streamer builds? 
(run 'vcgencmd version' to get firmware version)

You can update the firmware running sudo rpi-update. 

The numbers in that thread make more sense than what you're seeing here :-/

-usestills cannot reach 10fps. The most you will be getting is 8, probably more like 4-5, even if you ask for 10. 
So for video mode at 10fps vs usestills (at the same resolution and quality) I'd expect to see some difference in bitrate. 

If you didn't see a change enabling usestills then it either wasn't enabled or your measurements are off! 

If you got a 0.64Mb/s bitrate but still see 25% cpu use it kind of kills the theory of large bitrates causing high cpu.

Like your friend I only see a few percent cpu for 720p at 15fps. So I'd be interested to know what's going on here. 

Lighting will have a big difference on the image produced, I get a much yellower image under artificial light. It's possible to modify the image using exposure and whitebalance mode arguments.

nophead, 2016-09-20:
> Are they running the same rpi firmware and mjpg-streamer builds?

Yes they all give:
```
Mar 15 2016 14:51:23
Copyright (c) 2012 Broadcom
version 1bf9a9a77026af9128a339c82d72e331d3532ee4 (clean) (release)
```

>You can update the firmware running sudo rpi-update.

I ran ```sudo apt-get update``` and ```sudo apt-get upgrade``` before I attached the camera. I assume that amounts to the same thing?

>The numbers in that thread make more sense than what you're seeing here :-/

Yes I wish I could achieve that but have never been able to with the raspicam.

>-usestills cannot reach 10fps. The most you will be getting is 8, probably more like 4-5, even if you ask for 10. So for video mode at 10fps vs usestills (at the same resolution and quality) I'd expect to see some difference in bitrate.

On my oldest machine -usestills reduces the FOV without changing the exposure and dramatically  reduces the data rate from 20mbps to 320kbps. It does seem to have a very low frame rate but I lived with it because the picture quality is still good and the smaller FOV was actually better where the camera was mounted on the back of a Mendel90.

On my second machine -usestills doesn't affect the FOV, it darkens the exposure and makes it yellow and reduces the data rate up to 50%. Earlier today and last night it went from 25mbps to 12.5mbps but right now it is going from 20mbps to 15mbps. It doesn't seem to affect the frame rate though, that is still high. I am beginning to suspect that all it does is reduce the exposure and that has the knock on effect of reducing the data rate as darker scenes seem to compress better.

On my third machine the behavior changed. Last night it behaved exactly the same as machine number 2. Today though its maximum data rate seems to about 10mbps and this morning usestills had no effect. I just found I had a dodgy ethernet cable so it had actually fallen back to wifi and that was why it couldn't get past 10mbps. It is now behaving the same as machine number 2 again, including going yellow.

>If you didn't see a change enabling usestills then it either wasn't enabled or your measurements are off!

It is all repeatable except this morning -usestills didn't seem to affect the exposure on the third machine. It couldn't affect the data rate because it was capped by the wifi. I could have got confused this morning because do I have four OctoPrint tabs and four remote shells terminals.

My procedure is ```sudo nano /boot/octopt.txt```, run ```sudo service webcamd restart``` and then ```ps -ef | grep mjpg`` to verify the new command line. I then view the stream in my browser and look at the Windows Networking tab of Task Manager to see the data rate expressed as a percentage use of 1Gbps, which is my backbone rate.

>If you got a 0.64Mb/s bitrate but still see 25% cpu use it kind of kills the theory of large bitrates causing high cpu.

Well it does seem roughly proportional. The only machine I get low CPU with is the old one that has low frame rate with usestills.

>Like your friend I only see a few percent cpu for 720p at 15fps. So I'd be interested to know what's going on here.

I will see if I can get some data rate figures from him. What data rate are you seeing for those settings?

>Lighting will have a big difference on the image produced, I get a much yellower image under artificial light. It's possible to modify the image using exposure and whitebalance mode arguments.

Yes I realise that but -usestills on my second two machines makes even daylight very yellow and doesn't seem to do anything else. I am better off using quality to reduce the data on those.

I will try some more experiments as I have another camera and another RPI B.

Is there any firmware in the actual camera head? My first machine has a Noir version of the camera that is older than the other three cameras that I bought together. It must have a much older version of mjpg-streamer as well. I will try and find the git version but it seems a bit confused as I think I moved some files around.

I wonder what effect focus has on the ability to compress. My first machine has been focused on the centre of the bed but my other two cameras are still focused on infinity and see a much more complicated image of my untidy lab. The problem seems to be that my cameras produce far too much data so the image is not being compressed as well as it should be.

jacksonliam, 2016-09-20:
Thanks for the detail, it makes things a lot clearer. The data cap from wifi makes sense. 

apt-get upgrade will upgrade your firmware but only upto the latest raspbian release, rpi-update will get you the very latest firmware. But I don't think it's required here, the firmware you have should be fine. I'm not aware of any recent changes in this area. 

The encoder is in the GPU so shouldn't be affected by the actual camera hardware. NOIR will obviously produce a different picture to the normal camera though. 
You may well be right about the darker picture or more complex picture and bitrate. Also darker with usestills means longer exposure and lower framerate.

jacksonliam, 2016-09-20:
OK I think I have been able to reproduce all of your behaviour.
FYI one red herring may be that mjpg-streamer doesn't use much CPU when its not connected to a client (e.g. forgetting to refresh the browser page after restarting the server).

The CPU use issue really does seem to be data-rate related, but more dependant on framerate. 648x480 4fps with 85 quality in video mode should keep you ~8% on a pi1. 
As i previously thought there is no major difference in CPU usage between video and still modes when using the same resolution, quality and framerate.

Unlike I previously thought, the -quality argument works fine for modifying the bitrate, so I won't bother to add a seperate control for it.

My old test rigs with pre-pixel binning firmware do give a cropped image for video mode. This is not the case for the new firmware which uses the full sensor in both modes. I'll consider implementing cropping but its not high up my todo list.

The yellow image is the only thing I couldn't reproduce, I do see a variation in white balance when restarting mjpg-streamer but nothing that looks yellow. It might just be your pi snapping stills too fast, try slow it down with e.g. fps 4 (on latest firmware I only get 5.5. fps when specifying usestills with fps 10). 

In summary - I don't think there's any bug here I can fix. Use a low framerate to keep CPU use low, usestills is not required.

nophead, 2016-09-21:
I think what has changed is that on older versions usestills kept the same exposure and failed to maintain the frame rate, so the date rate went right down. The newer versions try to keep the target frame rate by reducing the exposure and so gets darker, which has the effect of reducing the data rate but nowhere near as much. If I set -fps lower then the image gets lighter. It still seems to have a yellow cast though.

What actually is the difference between a sequence of stills and video mode? I thought mjpg is just a sequence of compressed frames anyway.

How are you measuring the actual frame rate achieved?

648x480 4fps with 85 still uses 16% CPU or 20% when going through haproxy, so I don't seem to get the same performance as you and nowhere near as good as a USB camera which can run at 10 fps with a much lower data rate and CPU overhead.

nophead, 2016-09-21:
Another issue I have is that when connected with WIFI I get two kworker threads using a lot of CPU as well when streaming video. So an RPI camera streaming video over wifi just doesn't seem possible at any reasonable frame rate or quality.

jacksonliam, 2016-09-21:
The difference between stills and video mode is how the MMAL API is used. Stills mode asks for it to take a still JPEG picture, waits for a delay (based on the framerate passed to mjpg-streamer) and then asks for another one. The actual framerate is averaged over 100 frames and output to the terminal (probably in the log you referenced earlier). 
Video mode sets up a buffer and asks to capture video at the specified framerate and pipe it through the GPUs mjpeg encoder, I get callbacks  to transfer each frame from the buffer. 

I see the kworker threads for WiFi too, but also see those in stills mode. I'm not sure what the kernel is doing that's taking up so much CPU... Is it perhaps the wifi driver?

nophead, 2016-09-21:
Yes the kworker threads are definitely due to the wifi driver. I don't get them with a direct connection.

nophead, 2016-09-21:
My friend runs his RPI camera at 10 fps, everything else defaults and he only gets a 6.9mbps stream and uses 20% CPU with a wired ethernet connection. So he seems to be able to get about 4 times better compression than me and is able to stream about 10 times faster for the same CPU.

nophead, 2016-09-22:
I swapped the Wifi adaptor from a DIGITAZZ to Edimax EW-7811UN that seems to be recommended for RPI. I no longer get the kworker threads eating up CPU. Also I can set -fps 10 and default quality without the data rate going up to 25mbps and without using much CPU. **BUT** I get a 10 second lag on the video and a low frame rate displayed. I don't know where it manages to buffer up all that data on a RPI.

So it seems that the wifi adapter can throttle the data rate masking the problem. It may also depend on the browser and PC that is displaying the web page. The faster your network and client is the more load it puts on the RPI.

jacksonliam, 2016-09-22:
Ha, I have both those adapters too and was using the DIGITAZZ (it was like £3 from ebay). 

The http server will skip frames it doesn't have time to send to the client. When it finishes sending one frame it waits for the next one to be ready, skipping any frames that came from the camera during the sending of the previous frame. 

The camera always runs at a fixed framerate (even when no client is connected). 

A few MB of socket/network buffers should be enough to queue up 10 seconds at a few frames per second.

nophead, 2016-09-22:
Setting up a closer router ups the data rate and gets rid of the lag.
```-fps 4 -quality 30``` sends about 800kbps and uses about 20% CPU. That seems to be the best compromise. 

Viewing it twice in another browser window ups the CPU load to 30% so it seems most of the work is done in the web serving rather than processing the camera.

npasparuhov, 2017-03-04:
I notice something weird, even with -f/-fps 5 or -f/-fps 20 the camera stream with maximum frames, in my case the camera allows 30 fps/s and it just go every time with 30 fps ?!?!

jacksonliam, 2017-03-04:
@npasparuhov could you make a new issue as it's not really related to this. Also in the new issue include what hardware you used and the exact command you ran. Cheers.