jacksonliam/mjpg-streamer #44: mjpg-streamer compile issue

MrDini123, 2016-09-25:
Hi!

I tried to compile it on an embedded 2.6.31.8 kernel based nas, with libjpeg and v4l installed, but it says the following:

```
 53%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/input_uvc.c.o
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:90:13: error: 'V4L2_CID_POWER_LINE_FREQUENCY_AUTO' undeclared here (not in a function)
   { "auto", V4L2_CID_POWER_LINE_FREQUENCY_AUTO }
             ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c: In function 'cam_thread':
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:569:17: error: 'V4L2_CID_ROTATE' undeclared (first use in this function)
     V4L_INT_OPT(V4L2_CID_ROTATE, rot, "rotation")
                 ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:553:35: note: in definition of macro 'V4L_OPT_SET'
       if (input_cmd(pcontext->id, vid, IN_CMD_V4L2, settings->var, NULL) != 0) {\
                                   ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:569:5: note: in expansion of macro 'V4L_INT_OPT'
     V4L_INT_OPT(V4L2_CID_ROTATE, rot, "rotation")
     ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:569:17: note: each undeclared identifier is reported only once for each function it appears in
     V4L_INT_OPT(V4L2_CID_ROTATE, rot, "rotation")
                 ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:553:35: note: in definition of macro 'V4L_OPT_SET'
       if (input_cmd(pcontext->id, vid, IN_CMD_V4L2, settings->var, NULL) != 0) {\
                                   ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:569:5: note: in expansion of macro 'V4L_INT_OPT'
     V4L_INT_OPT(V4L2_CID_ROTATE, rot, "rotation")
     ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:569:17: warning: passing argument 2 of 'input_cmd' makes integer from pointer without a cast
     V4L_INT_OPT(V4L2_CID_ROTATE, rot, "rotation")
                 ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:553:35: note: in definition of macro 'V4L_OPT_SET'
       if (input_cmd(pcontext->id, vid, IN_CMD_V4L2, settings->var, NULL) != 0) {\
                                   ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:569:5: note: in expansion of macro 'V4L_INT_OPT'
     V4L_INT_OPT(V4L2_CID_ROTATE, rot, "rotation")
     ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:96:5: note: expected 'unsigned int' but argument is of type 'const struct <anonymous> *'
 int input_cmd(int plugin, unsigned int control, unsigned int group, int value, char *value_string);
     ^
plugins/input_uvc/CMakeFiles/input_uvc.dir/build.make:77: recipe for target 'plugins/input_uvc/CMakeFiles/input_uvc.dir/input_uvc.c.o' failed
make[2]: *** [plugins/input_uvc/CMakeFiles/input_uvc.dir/input_uvc.c.o] Error 1
CMakeFiles/Makefile2:255: recipe for target 'plugins/input_uvc/CMakeFiles/input_uvc.dir/all' failed
make[1]: *** [plugins/input_uvc/CMakeFiles/input_uvc.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
```

Many thanks!

jacksonliam, 2016-09-25:
Hmmm which version of v4l do you have?

MrDini123, 2016-09-25:
I have old git 20110614 version, because this is the last release which supports uvcvideo with this old kernel.

jacksonliam, 2016-09-25:
Hmmm, It might be possible to conditionally #define out the lines its complaining about based on v4l version, but Its not work I'm going to do, mainly because I don't have anything with a kernel like that to test on.

For your purposes, edit  mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c
and delete lines 
90:
` { "auto", V4L2_CID_POWER_LINE_FREQUENCY_AUTO }`
569
` V4L_INT_OPT(V4L2_CID_ROTATE, rot, "rotation")`

Should work without those as long as you don't try to use those options.

MrDini123, 2016-09-25:
Yes, it worked.

Another error now...:

```
[ 73%] Building C object plugins/output_file/CMakeFiles/output_file.dir/output_file.c.o
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/output_file/output_file.c: In function 'output_init':
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/output_file/output_file.c:503:28: error: 'V4L2_CTRL_TYPE_STRING' undeclared (first use in this function)
  filename_ctrl.ctrl.type = V4L2_CTRL_TYPE_STRING;
                            ^
/i-data/bf835951/build/mjpg-streamer/mjpg-streamer-experimental/plugins/output_file/output_file.c:503:28: note: each undeclared identifier is reported only once for each function it appears inplugins/output_file/CMakeFiles/output_file.dir/build.make:54: recipe for target 'plugins/output_file/CMakeFiles/output_file.dir/output_file.c.o' failed
make[2]: *** [plugins/output_file/CMakeFiles/output_file.dir/output_file.c.o] Error 1
CMakeFiles/Makefile2:305: recipe for target 'plugins/output_file/CMakeFiles/output_file.dir/all' failed
make[1]: *** [plugins/output_file/CMakeFiles/output_file.dir/all] Error 2
Makefile:117: recipe for target 'all' failed
make: *** [all] Error 2
```

jacksonliam, 2016-09-25:
Do you plan to use the file output plugin? 

You can probably delete that line too (mjpg-streamer/mjpg-streamer-experimental/plugins/output_file/output_file.c line 503) 

But I'm not sure what in that plugin will break without looking tracing it properly (on phone atm)

MrDini123, 2016-09-25:
No, how Can I disable it?

And one more thing! After the cmake it said, the http managment is disabled... So how Can enable it manualy?

Thanks!

jacksonliam, 2016-09-25:
Did removing the line not work? 
I think you can just remove 'add_subdirectory(plugins/output_http)' from CMakeLists.txt 

HTTP management is an experimental feature to allow you to change properties (e.g. framerate)  on the fly. I'm not sure it's working now (no one has worked on it for a long time) so it's disabled now. It's seperate from the streaming feature (output_http) 

Cheers, Liam.

MrDini123, 2016-09-26:
I don't tried the line removing. I thought remove the whole module, if it is unneded and not works on my nas...

But now, I am trying to compile a newer v4l and look what happens.

MrDini123, 2016-09-26:
Thanks for Your help!

With the media build headers and with my kernel headers specified at the cmake command, I Could built it without uncommenting lines from source. :)

Cool, will try it with a camera tomorrow!

closed 2016-09-26 by MrDini123

MrDini123, 2016-09-27:
Now, I could try it with my cameras. Works nicely. And I surprised, because I Cannot see the mjpg at the top's output... With motion, I had ~25% CPU busy.... With one cam. Nice!

Is it works with gspca drivers too?

Thanks a lot!