jacksonliam/mjpg-streamer #45: output_file plugin no longer recognizes --command/-c option

ducksauz, 2016-09-29:
I'm starting it with:
`
-o "output_file.so -d 10000 -f $PICSDIR/tmp --command move-pic.sh"
`

and get this error:
`FILE output plugin: unrecognized option '--command'
 ---------------------------------------------------------------
 Help for output plugin..: FILE output plugin
 ---------------------------------------------------------------
 The following parameters can be passed to this plugin:

 [-f | --folder ]........: folder to save pictures
 [-m | --mjpeg ].........: save the frames to an mjpg file
 [-d | --delay ].........: delay after saving pictures in ms
 [-i | --input ].........: read frames from the specified input plugin
 The following arguments are takes effect only if the current mode is not MJPG
 [-s | --size ]..........: size of ring buffer (max number of pictures to hold)
 [-e | --exceed ]........: allow ringbuffer to exceed limit by this amount
 [-c | --command ].......: execute command after saving picture
 ---------------------------------------------------------------`

It even taunts me with the option in the help it's trying to give.

When I look at the source for the plugin, it looks like the command option got lost from the getopt definition.

https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_file/output_file.c#L362-L382

jacksonliam, 2016-09-29:
Yep you're right, I can't see why it was removed. If it was accidental or not. But all the code to make it work still seems to be there. 

Are you able to do a pull request?

ducksauz, 2016-09-30:
Unfortunately, I can't do a pull request. I'm subject to OSS contribution restrictions at work. :/

jacksonliam, 2016-09-30:
No problem, I'll leave this open here but I wouldn't expect me to get to it anytime soon!

fluller, 2017-09-08:
I met the same issue, i'll do a pull request.

PR #90 (2018-01-25): Fix the missing '-c command' option
Fixes #45
commit b26f484c (2018-01-25): Fix the missing '-c command' option (#90)

Fixes #45
closed 2018-01-25 by jacksonliam