jacksonliam/mjpg-streamer #46: Rotate input_uvc

ian2000611, 2016-10-02:
I would like it if input uvc could be patched to support the -rot option from input_raspi

If this is a software rotation and can be applied that would be fantastic. If it is a configuration parameter sent to the picam hardware , do you know of a plugin that well do this on a logitech C270?

jacksonliam, 2016-10-02:
There is a rot parameter for input uvc already. It's up to your hardware to support it. The raspi one is done in the GPU, I think it may be too expensive of an operation to do in the CPU.

If your camera doesn't support it, perhaps do the rotation on the receiving side. It might be possible to set something in the EXIF data so that's done automatically, but it's not something I'm going to implement!

closed 2016-10-02 by jacksonliam

your-friend-alice, 2023-07-17:
any chance you'd reopen this now that the [image-orientation](https://developer.mozilla.org/en-US/docs/Web/CSS/image-orientation) css property exists and makes use of EXIF?