jacksonliam/mjpg-streamer #47: Duplicate last frame in case the input is not available

mad-ady, 2016-11-01:
Here's my problem. I'm reading a jpeg stream from a UVC webcam but some of the frames are corrupt (sidenote: it seems it's not the camera's fault, it's related to the USB driver in my Odroid C2). Corrupted frames usually have missing EOI or other corruptions. Here's an example when reading from mjpg_streamer with ffmpeg:

[mjpeg @ 0x4876a0] mjpeg: unsupported coding type (cd)
[mjpeg @ 0x4876a0] EOI missing, emulating
[mjpeg @ 0x4876a0] mjpeg: unsupported coding type (c5)
[mjpeg @ 0x4876a0] EOI missing, emulating
[mjpeg @ 0x4876a0] EOI missing, emulating

In case mjpg_streamer decodes the frames before pushing them in the stream I'd like you to add the feature to send the last valid frame if the current frame can't be decoded.

Although ffmpeg happily consumes the stream, it doesn't duplicate missed frames and this results in dropping the framerate.

If mjpg_streamer does not decode the frame at all, then you can't fix this issue at your end...

Thanks

jacksonliam, 2016-11-01:
The jpegs are unfortunately not decoded so I don't know if they're corrupted. 

I guess you don't need to decode the whole jpeg to tell if it's corrupted? I'd take a pull request of a feature (on a flag, disabled by default) which rejected frames from the camera that don't pass a sanity check. But it's not something I'll be doing myself. 

IIRC someone implanted a 'skip' feature for some cameras that produce corrupted frames, so if it's every nth frame you could try that!

mad-ady, 2016-11-03:
I understand. I tried to fix it from ffmpeg (by having it output a constant framerate by duplicating dropped frames), but so far I have been unsuccessful. The only way I managed to do it was to write mjpg_streamer output into a fake webcam (set up with v4l2loopback) and read the input from there. The loopback duplicates missing frames and all is well.

You may close this request - there is no need to start decoding the frames just for this feature - it would needlessly consume CPU cycles.

Thank you.

closed 2016-11-03 by mad-ady