jacksonliam/mjpg-streamer #49: undefined symbol: _ZN2cv12VideoCaptureC1Ev

jakinto, 2016-12-07:
Hi,

I'm trying to execute this command to stream webcam with low quality to http:

`mjpg_streamer -i 'input_opencv.so -d /dev/video0 -q 50' -o 'output_http.so -w ./www'`

And I'm getting this error:

`MJPG-streamer [15]: starting application
MJPG Streamer Version.: 2.0
MJPG-streamer [15]: MJPG Streamer Version.: 2.0

mjpg_streamer: symbol lookup error: ./input_opencv.so: undefined symbol: _ZN2cv12VideoCaptureC1Ev`

Other inputs work fine. This must be related with OpenCV. Any ideas? Which parameters do you use when installing OpenCV?

Thanks in advance.

jacksonliam, 2016-12-07:
Which version of opencv have you got installed?

jakinto, 2016-12-09:
OpenCV 3.1

virtuald, 2016-12-09:
Try this commit: https://github.com/robotpy/mjpg-streamer/commit/d558daa5b96a42ef18399cb857639e94fff06c2d

I've been meaning to PR it, but I haven't had a chance to actually test it on my Fedora machine to make sure it still works.

jakinto, 2016-12-09:
Now I'm getting:

```
MJPG-streamer [781]: starting application
MJPG Streamer Version.: 2.0
MJPG-streamer [781]: MJPG Streamer Version.: 2.0

 i: device........... : /dev/video0
MJPG-streamer [781]: device........... : /dev/video0

 i: Desired Resolution: 640 x 480
MJPG-streamer [781]: Desired Resolution: 640 x 480

 i: VideoCapture::open() failed
MJPG-streamer [781]: VideoCapture::open() failed
```

Any ideas?

virtuald, 2016-12-09:
Presume that you've checked that /dev/video0 exists? It might not support that resolution... there's a v4l tool you can use to query the webcam, I've forgotten what it's called...

jakinto, 2016-12-10:
Using this command works:

`mjpg_streamer -i 'input_uvc.so -d /dev/video0' -o 'output_http.so -w ./www'`

So I don't think is related with the webcam.

jakinto, 2016-12-12:
The problem was **-d** param. Remove this parameter to use the default camera:

`mjpg_streamer -i 'input_opencv.so -q 70' -o 'output_http.so -w ./www'`

closed 2016-12-12 by jakinto

virtuald, 2016-12-12:
Oh right! This must be related to https://github.com/opencv/opencv/pull/6391 -- if you patch your OpenCV installation with that, it should work. For some reason I thought that didn't apply to things like `/dev/video0`...