jacksonliam/mjpg-streamer #53: Doesn't work.

ambrosa, 2016-12-24:
I'm running a fresh installation of latest Raspbian Jessie Lite in my Raspberry PI2
I've an external USB webcam connected to it. It's a :
```
[    6.678754] Linux video capture interface: v2.00
[    6.736158] usb 1-1.5: 3:1: cannot get freq at ep 0x84
[    6.770614] usbcore: registered new interface driver snd-usb-audio
[    6.772477] uvcvideo: Found UVC 1.00 device Trust Webcam (0c45:6340)
[    6.792758] input: Trust Webcam as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/input/input0
[    6.793149] usbcore: registered new interface driver uvcvideo
[    6.793154] USB Video Class driver (1.1.1)

```
It works well with other Raspbian sw like 'fswebcam' (and other).
But I need a light web streamer software and mjpeg-streamer works fine. But NOT YOURS :-(
I've made first installation reading this guide: http://www.acmesystems.it/video_streaming
I've downloaded their (old) source http://terzo.acmesystems.it/download/webcam/mjpg-streamer.tar.gz , I've compiled it without any problem and it works fine (see a screnshoot https://postimg.org/image/3mit854xj/   webcam image is dark because room light are switched off. Webcam works fine).
I test it with:
```
root@rpi2:~/mjpg-streamerl# export LD_LIBRARY_PATH="$(pwd)"
root@rpi2:~/mjpg-streamer# ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -y -f 2 -r 640x480" -o "output_http.so -p 7011 -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 2
 i: Format............: YUV
 i: JPEG Quality......: 80
Adding control for Pan (relative)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Tilt (relative)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Pan Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Tilt Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Pan/tilt Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Focus (absolute)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
mapping control for Pan (relative)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Tilt (relative)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Pan Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Tilt Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Pan/tilt Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Focus (absolute)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for LED1 Mode
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for LED1 Frequency
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Disable video processing
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Raw bits per pixel
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
 o: www-folder-path...: ./www/
 o: HTTP TCP port.....: 7011
 o: username:password.: disabled
 o: commands..........: enabled

```

After that I've discovered your newer version. Downloaded and compiled without problem.
Run it:
```
root@rpi2:~/mjpg-streamer-master/mjpg-streamer-experimental# export LD_LIBRARY_PATH="$(pwd)"
root@rpi2:~/mjpg-streamer-master/mjpg-streamer-experimental# ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -y -f 2 -r 640x480" -o "output_http.so -p 7010 -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 2
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 2 to 5
 i: Frame period time ......: 500 ms
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path...: ./www/
 o: HTTP TCP port.....: 7010
 o: username:password.: disabled
 o: commands..........: enabled
```
Missing "Adding control" and "mapping control"
No image is shown into web interface. It doesn't work.

If I force the LD_LIBRARY_PATH to old version, it works:
```
root@rpi2:~/mjpg-streamer-master/mjpg-streamer-experimental# export LD_LIBRARY_PATH=/root/mjpg-streamer
root@rpi2:~/mjpg-streamer-master/mjpg-streamer-experimental# ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -y -f 2 -r 640x480" -o "output_http.so -p 7010 -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 2
 i: Format............: YUV
 i: JPEG Quality......: 80
Adding control for Pan (relative)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Tilt (relative)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Pan Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Tilt Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Pan/tilt Reset
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
Adding control for Focus (absolute)
UVCIOC_CTRL_ADD - Error: Inappropriate ioctl for device
mapping control for Pan (relative)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Tilt (relative)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Pan Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Tilt Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Pan/tilt Reset
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Focus (absolute)
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for LED1 Mode
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for LED1 Frequency
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Disable video processing
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
mapping control for Raw bits per pixel
UVCIOC_CTRL_MAP - Error: Inappropriate ioctl for device
 o: www-folder-path...: ./www/
 o: HTTP TCP port.....: 7010
 o: username:password.: disabled
 o: commands..........: enabled
```
It works !

If you need further debug, let me know.

Thanks and ... Merry Christmas.

ambrosa, 2016-12-24:
Added "-n" to input : non success again.
```
root@rpi2:~/mjpg-streamer-master/mjpg-streamer-experimental# ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -y -f 2 -n -r 640x480" -o "output_http.so -p 7010 -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 2
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 2 to 5
 i: Frame period time ......: 500 ms
 o: www-folder-path...: ./www/
 o: HTTP TCP port.....: 7010
 o: username:password.: disabled
 o: commands..........: enabled
```

jacksonliam, 2016-12-24:
Are you seeing a black frame or no frame at all? 

Can you try running with an fps of 5 rather than 2?

Cheers, Liam.

mnoorenberghe, 2016-12-24:
I'm having the same issue on a fresh Raspbian Jessie Lite Pi3 install. On this same Pi everything was working fine until recently so that's why I tried a fresh install. I would guess that some package update broke this as I didn't do other major changes since this Pi's primary purpose is to run mjpg-streamer.

> Are you seeing a black frame or no frame at all? What happens if you shine a light at the camera?

For me the requests for images/videos never get a response. http://raspberrypi3-03.local:10088/?action=snapshot

```
curl -vvvv http://raspberrypi3-03.local:10088/?action=snapshot
*   Trying fe80::6dae:f713:5a85:69bb...
* Connected to raspberrypi3-03.local (fe80::6dae:f713:5a85:69bb) port 10088 (#0)
> GET /?action=snapshot HTTP/1.1
> Host: raspberrypi3-03.local:10088
> User-Agent: curl/7.49.1
> Accept: */*
> 
```
… and it hangs here.

> What happens if you shine a light at the camera?

No difference.

> Can you try running with an fps of 5 rather than 2?

That didn't help for me.

System details:
Linux raspberrypi3-03 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l GNU/Linux
All packages up-to-date on Jessie. I only installed `git`, `cmake`, `libjpeg8-dev` and dependencies on top of Jessie Lite and configured some stuff with raspi-config and modified wpa_supplicant.conf. I didn't make other changes to the fresh install.

mnoorenberghe, 2016-12-24:
And I just installed fswebcam and it works perfectly.

mnoorenberghe, 2016-12-24:
The `input_uvc` thread backtrace:
```
(gdb) bt full
#0  0x76e1df28 in ioctl () at ../sysdeps/unix/syscall-template.S:81
No locals.
#1  0x76d3b054 in xioctl (fd=4, IOCTL_X=-1069263343, arg=0x8a7974)
    at /home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c:80
        ret = 0
        tries = 4
#2  0x76d3d488 in uvcGrab (vd=0x8a7830) at /home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c:560
        ret = 0
#3  0x76d393a4 in cam_thread (arg=0x245b8 <global+8>)
    at /home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c:631
        __cancel_buf = {__cancel_jmp_buf = {{__cancel_jmp_buf = {0, 0, 2128367704, 338, 9074360, 1995510640, 0, 1988812380, 936754963, 
                931979515, 0 <repeats 54 times>}, __mask_was_saved = 0}}, __pad = {0x768ade70, 0x0, 0x0, 0x0}}
        __cancel_routine = 0x76d39d14 <cam_cleanup>
        __cancel_arg = 0x245b8 <global+8>
        __not_first_call = 0
        in = 0x245b8 <global+8>
        pcontext = 0x8a76b0
        settings = 0x0
        every_count = 0
        quality = 95
        __FUNCTION__ = "cam_thread"
#4  0x76eaee90 in start_thread (arg=0x768ae460) at pthread_create.c:311
        pd = 0x768ae460
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {1988814024, 0, 2128367704, 338, 9074360, 1995510640, 0, 1988812700, 936753267, 
                934643827, 0 <repeats 54 times>}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {prev = 0x0, 
              cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#5  0x76e25598 in ?? () at ../ports/sysdeps/unix/sysv/linux/arm/nptl/../clone.S:92 from /lib/arm-linux-gnueabihf/libc.so.6
No locals.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

ambrosa, 2016-12-25:
mnoorenberghe  , I don't use GDB since maaaany years :-)
Can you give me details how you have obtained your backtrace ? So I can backtrace me too and compare my results with yours.

ambrosa, 2016-12-25:
Dear jacksonliam,

same issue here as mnoorenberghe:
- no response from server (it looks hang)
- changing fps doesn't work

jacksonliam, 2016-12-26:
Can anyone tell me if this is due to a change in the newer raspbian or a change to this repo?

ariovistus, 2017-02-03:
I'm using a build I built last spring for raspbian jessie, I want to say full, and now running on raspian lite and getting:

```
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 320 x 240
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path...: /usr/local/share/mjpg-streamer/www/
 o: HTTP TCP port.....: 5800
 o: username:password.: disabled
 o: commands..........: enabled

```

so I think it's either raspbian changed, or I'm missing some config

jacksonliam, 2018-01-25:
@ambrosa, @mnoorenberghe I pushed a fix for another issue #100 which looks like your issue, can you try the latest code (a `git pull` then a `make`) and see if you have any more luck?

mnoorenberghe, 2018-01-26:
Unfortunately it didn't seem to make a difference:
```
$ curl -vvvv http://raspberrypi3-03.local:10088/?action=snapshot
* Hostname was NOT found in DNS cache
*   Trying 192.168.0.30...
* Connected to raspberrypi3-03.local (192.168.0.30) port 10088 (#0)
> GET /?action=snapshot HTTP/1.1
> User-Agent: curl/7.38.0
> Host: raspberrypi3-03.local:10088
> Accept: */*
> 
```

[streamer_log.txt](https://github.com/jacksonliam/mjpg-streamer/files/1666934/streamer_log.txt)

jacksonliam, 2018-01-26:
Thanks for checking! Sorry it still doesn't work. I think it is a problem with the queued buffers but not sure what. 

Is the stack trace always stuck on xioctl / uvcgrab?

Is there anything useful in dmesg? 

If you leave it running for some time (e.g. 10 mins) does mjpg-streamer output any errors/timeouts? 

Cheers, Liam.

DrLous, 2018-03-23:
Hi,

Do you try without the fps option ? and with the -e option (like -e 10 for starting) ?
Watch your cpu usage too :)

Br.

ighack, 2018-03-31:
I have the sam issue

qpens, 2018-04-04:
Hi, 
I'm getting an error with the compilation of older version. Did you use just make command or something else?

jacksonliam, 2018-04-14:
I have pushed changes I think fixed this (821c330) can you please all test the fixes?
e.g.
```
git pull
make distclean
make
```
(and if you ran it before)
```
sudo make install
```
Cheers.

Nossiac, 2018-04-17:
similiar error occurred here with the latest code you pushed:

./mjpg_streamer -i "input_uvc.so --fps 5" -o "output_http.so -p 8090"

```
 MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e923161b
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 5
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8090
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
```

I also noticed that my openwrt set works fine, it uses code from here:
[https://svn.code.sf.net/p/mjpg-streamer/code/mjpg-streamer-experimental](https://svn.code.sf.net/p/mjpg-streamer/code/mjpg-streamer-experimental)
revision r182

jacksonliam, 2018-04-17:
Thanks for testing, that's disappointing!

Do you get any more useful info now if you run with debugging enabled?

Same instructions as above but when running `make` instead run `make CMAKE_BUILD_TYPE=Debug`

Nossiac, 2018-04-18:
@jacksonliam 
Here it is, with debug on.  The video was fine actually, just found some error logs about "UVCIOC_CTRL_MAP".

```
MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e923161b
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, input_init(), 164): argv[0]=UVC webcam grabber
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, input_init(), 382): input id: 0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, input_init(), 431): vdIn pn: 0
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 130): Input name: Camera 1
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 143): Current size: 640x480
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 172): Supported format: YUV 4:2:2 (YUYV)
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 640x480
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 160x120
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 176x144
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 320x240
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 352x288
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 640x360
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 640x400
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 768x480
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 800x456
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 800x504
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 800x600
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 864x480
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 960x720
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 1280x720
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 1280x800
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 1600x904
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 1600x1000
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 205):      Supported size: 1600x1200
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 172): Supported format: MJPEG
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 640x480
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 160x120
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 176x144
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 320x240
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 352x288
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 640x360
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 640x400
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 768x480
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 800x456
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 800x504
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 800x600
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 864x480
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 960x720
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 1280x720
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c, init_videoIn(), 203):      Supported size with the current format: 1280x800
[... cut ...]
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 726): copying frame from input: 0
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 726): copying frame from input: 0
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 726): copying frame from input: 0
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 726): copying frame from input: 0
 DBG(/test/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 726): copying frame from input: 0
```

dazz100, 2018-04-18:
Hi
I have a fresh version of mjpg-streamer I grabbed a few hours ago.  I have the same problem.

```
 $ ./mjpg_streamer -i "input_uvc.so  -r 640x480  -d /dev/video0  -f 2 "   -o "output_http.so -p 8080 -w /usr/local/share/mjpg-streamer/www"
MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e923161b
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 2
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 2 to 5
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: /usr/local/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled

```

Doesn't appear to be listening to the brower.  I am using http://ip_address:8080/
```
$ netstat -anl | grep 8080
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN
tcp6       0      0 :::8080                 :::*                    LISTEN
```

The folder "/usr/local/share/mjpg-streamer/www/" is empty.

The web browser address is:  http://ip_address:8080/

The mjpg-streamer is running on a Pi on a local network connected to my main PC.

Nossiac, 2018-04-18:
@dazz100 was the video ok? I was using logic C905, the error log was seen but the video was ok. 

maybe my logic camera does support those ioctl options?

jacksonliam, 2018-04-18:
This issue is about the video stream not working. The UVCIOC_CTRL_MAP errors are harmless and I'm not attempting to fix those here. @Nossiac I believe the C905 works fine and you never had issues with the video?

@dazz100 I believe your issue is that you cannot access the web interface? mjpg-streamer appears to be listening... Why are you using "/usr/local/share/mjpg-streamer/www/" if it is empty? Use "./www"

@ambrosa or @mnoorenberghe any chance you can give the new commit a test with your problematic cameras?

Nossiac, 2018-04-18:
yes, the video is fine in my case. that the errors are harmless cures the anxiety, thanks

dazz100, 2018-04-18:
Hi

Thanks for the quick response.

I am using /usr/local/share/mjpg-streamer/www/ because I don't know what 
I should be using.  The documentation is missing some of the fine (but 
important) detail in places.

eg.  The instructions for git clone couldn't be found in the documentation.

git clone https://github.com/jacksonliam/mjpg-streamer.git

I am sure you know that but I had to go looking for it.

OK, I changed the command to point to .www  and it is working. Nice.

Thanks for the work you are doing.

Regards

Darren Conway
36 Orr Crescent
Lower Hutt
New Zealand
ph +64  (0)4 569 1963

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

mnoorenberghe, 2018-04-18:
> @ambrosa or @mnoorenberghe any chance you can give the new commit a test with your problematic cameras?

Yeah, I'll try find time to test it soon… unfortunately since I encountered this bug I use different software so testing this fix means shutting that down. I wanted to test on the weekend but it was really bad timing as the power at that location went out for almost 24 hours meaning I couldn't test.

BytEvil, 2018-04-26:
Same problem on my Logitech C910 Webcam. 
```
mjpg_streamer -i "input_uvc.so -r 1280x720 -f 60 -d /dev/video0" -o "output_http.so -p 8015 -w /www/cam.*****.ru"

MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e923161b
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: /www/cam.sama63.ru/
 o: HTTP TCP port........: 8015
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
libv4l2: error turning on stream: Message too long
Unable to start capture: Message too long
i: Error grabbing frames
```

jacksonliam, 2018-04-26:
@BytEvil - that looks different to what is being discussed here. Please create a new issue called "libv4l2 Message too long" with that information in.

lzytniak, 2019-10-30:
Hi,

I am using Waveshare OV5648 5MP USB Camera with auto focusing, when I try to run I get:
`~/mjpg-streamer-master/mjpg-streamer-experimental $ ./mjpg_streamer -i "./input_uvc.so -r 800x600" -o "output_http.so -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 800 x 600
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled`

tswaehn, 2021-02-07:
Hi Jack,
I am using a webcam on my RPI3b+ with raspbian-buster. Just installed uvcbiew with ssh-x-forwarding. video images works fine.

however mjpg-streamer refused to work. i have options to test/debug things. please let me know what I can do to root cause this issue. 

thanks in advance.

this is my cmd line and log

```
pi@apoc:/usr/local/lib/mjpg-streamer $ sudo mjpg_streamer -i "input_uvc.so -f 10 -y" -o 'output_http.so -w /var/www/html/webcam -l 0.0.0.0 -p 8888'
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: /var/www/html/webcam/
 o: HTTP TCP port........: 8888
 o: HTTP Listen Address..: 0.0.0.0
 o: username:password....: disabled
 o: commands.............: enabled

```
I think the blocking part might be this line here:

```
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
```

jacksonliam, 2021-02-07:
You can ignore the ioctl errors, they shouldn't affect anything to do with the stream.

What do you mean by refuse to work? Does it crash out at this point?
Can you get to the index page over http? Is the stream returning black frames or no frames? Etc.

Have you tried without the `-l 0.0.0.0` option?

tswaehn, 2021-02-07:
`Have you tried without the -l 0.0.0.0 option?`

yes, I tried 
- without
- localhost
- ip
- hostname
it doesnt work and always returns with

`http://apoc:8888/`

```
404: Not Found!
Could not open file
```

jacksonliam, 2021-02-07:
What is the contents of the /var/www/html/webcam directory you have specified?

tswaehn, 2021-02-07:
`What is the contents of the /var/www/html/webcam directory you have specified?`

its empty. what should be there?

tswaehn, 2021-02-07:
-- btw. thanks for your fast response :)

tswaehn, 2021-02-07:
oh man, I am sometimes that stupid. I should use 

`http://apoc:8888/?action=stream`

instead of:

`http://apoc:8888/`

which works perfectly fine.

jacksonliam, 2021-02-07:
Cool, sorted

closed 2021-02-07 by jacksonliam

tswaehn, 2021-02-07:
yes, thanks.

ProgrammerIsMe, 2021-12-15:
hi Jack, i have encountered the same problem on the latest version, and i use raspberry 3b+ and the camera with it. The local webpage could be open but there is no video frames. Here is the error:
MJPG Streamer Version: git rev: c92ae66e27533a5f80ba6ac3ecfe2351337681fa
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 5
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
Thanks in advance!

lida2003, 2024-01-22:
Ho....... long list, is it fixed? It seems quite a long time.

vinibali, 2024-10-03:
the ioctl related errors are not very relevant, I just came here because of those btw.
the mjpg-streamer works fine along with those errors IIRC the
```
-n
```
flag turns off the ioctl related probes.