jacksonliam/mjpg-streamer #57: Cannot "make"

stefanLeo, 2017-01-07:
pi@leopi:~/mjpg-streamer-experimental/mjpg-streamer/mjpg-streamer-experimental $ make mjpg_streamer input_file.so input_uvc.so input_raspicam.so output_http.so
cc     mjpg_streamer.c   -o mjpg_streamer
/tmp/ccEbrHFM.o: In function `signal_handler':
mjpg_streamer.c:(.text+0x3b8): undefined reference to `dlclose'
mjpg_streamer.c:(.text+0x4ac): undefined reference to `dlclose'
/tmp/ccEbrHFM.o: In function `main':
mjpg_streamer.c:(.text+0xa10): undefined reference to `daemon_mode'
mjpg_streamer.c:(.text+0xe78): undefined reference to `dlopen'
mjpg_streamer.c:(.text+0x1084): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1118): undefined reference to `dlsym'
mjpg_streamer.c:(.text+0x1180): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1210): undefined reference to `dlsym'
mjpg_streamer.c:(.text+0x1278): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1308): undefined reference to `dlsym'
mjpg_streamer.c:(.text+0x1370): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1400): undefined reference to `dlsym'
mjpg_streamer.c:(.text+0x17f8): undefined reference to `dlopen'
mjpg_streamer.c:(.text+0x1a1c): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1ab4): undefined reference to `dlsym'
mjpg_streamer.c:(.text+0x1b24): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1bb8): undefined reference to `dlsym'
mjpg_streamer.c:(.text+0x1c20): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1cb4): undefined reference to `dlsym'
mjpg_streamer.c:(.text+0x1d1c): undefined reference to `dlerror'
mjpg_streamer.c:(.text+0x1db0): undefined reference to `dlsym'
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'mjpg_streamer' failed
make: *** [mjpg_streamer] Error 1

I executed the installation process as described in here - simply changing the GIT URL.
Using latest RASPBIAN IMAGE.

THX

jacksonliam, 2017-01-07:
Not seeing this, what do you mean by "I executed the installation process as described in here - simply changing the GIT URL."?

jacksonliam, 2017-01-15:
Could not reproduce on lastest raspbian. The above is not the output you get from building this repo, so I assume you're trying to build something else.

I suggest you delete what you have, `git clone https://github.com/jacksonliam/mjpg-streamer.git` and follow the build instructions as given https://github.com/jacksonliam/mjpg-streamer

closed 2017-01-15 by jacksonliam

yuyat, 2018-10-21:
Instalation problem on latest rasbrian image.

 make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
make -C _build
make[1]: se entra en el directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[2]: se entra en el directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se entra en el directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se sale del directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se entra en el directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[  4%] Building C object CMakeFiles/mjpg_streamer.dir/mjpg_streamer.c.o
[  8%] Building C object CMakeFiles/mjpg_streamer.dir/utils.c.o
[ 12%] Linking C executable mjpg_streamer
make[3]: se sale del directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 12%] Built target mjpg_streamer
make[3]: se entra en el directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se sale del directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[3]: se entra en el directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
[ 16%] Building C object plugins/input_file/CMakeFiles/input_file.dir/input_file.c.o
[ 20%] Linking C shared library input_file.so
/usr/bin/ld: /usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libc.a(dl-tsd.o)(.text+0x14): la reubicación R_ARM_TLS_LE32 no se permite en objetos compartidos
/usr/lib/gcc/arm-linux-gnueabihf/6/../../../arm-linux-gnueabihf/libc.a(dl-tsd.o): En la función `_dl_initial_error_catch_tsd':
(.text+0x14): reubicación peligrosa: no se admite la reubicación
collect2: error: ld returned 1 exit status
plugins/input_file/CMakeFiles/input_file.dir/build.make:94: fallo en las instrucciones para el objetivo 'plugins/input_file/input_file.so'
make[3]: *** [plugins/input_file/input_file.so] Error 1
make[3]: se sale del directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
CMakeFiles/Makefile2:122: fallo en las instrucciones para el objetivo 'plugins/input_file/CMakeFiles/input_file.dir/all'
make[2]: *** [plugins/input_file/CMakeFiles/input_file.dir/all] Error 2
make[2]: se sale del directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:127: fallo en las instrucciones para el objetivo 'all'
make[1]: *** [all] Error 2
make[1]: se sale del directorio '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:17: fallo en las instrucciones para el objetivo 'all'
make: *** [all] Error 2