jacksonliam/mjpg-streamer #58: NO_LIBJPEG and YUV mode

alexscarbro, 2017-01-08:
Hello,
I've just stumbled across an issue where when using the -y (YUV) option for the input_uvc module it rejected this as a syntax error. This was due to the LIBJPEG not being installed. However, "-y" was still being shown as a valid command line switch, despite LIBJPEG not being installed.

Perhaps the "-y" switch should not be available when the binary is built without LIBJPEG?

Thanks,
Alex

jacksonliam, 2017-01-08:
Perhaps, but running without libjepg is not something I really want to encourage. I'll leave this issue here incase I'm making changes to that area and can simply include some extra #defines

Cheers!

alexscarbro, 2017-01-08:
In retrospect, I clearly wasn't a good idea (building & running without libjpeg), but at the time of doing the build I hadn't realised it wasn't installed. Its more to save some other poor bugger from going around in circles whilst they figure out why the YUV option appears in the --help, but gets thrown out at runtime as an invalid flag.

Great work by the way, I'm running it on an Orange Pi Zero along with LaserWeb3 - it works beautifully!

roger-, 2017-01-10:
libjpg may not be available on OpenWrt systems, for example, so I think it'd be good to work properly when it's not available.

jacksonliam, 2017-01-15:
Hey, what steps are you taking to build without libjpeg? It should show help and exit if you try to use `-y` when building without. Makes me think the `#define` `NO_LIBJPEG` isn't being set...

I will adjust the help though.

devekrehbiel, 2017-08-12:
Is there a way to use a webcam that does not support MPEG and requires the -y flag that does not cause a massive surge in CPU resources?  Most of the elcheapo webcams require the -y but wow, does that ever slow the stream down. This is the command line I am using...
/usr/local/bin/mjpg_streamer -b -i "/usr/local/lib/input_uvc.so -d /dev/video0 -r 640x480 -f 30 -y" -o "/usr/local/lib/output_http.so -p 9000 -w /usr/local/www"

This is an awesome program and my compliments to those involved! If only I could get html buttons to work with my pan/tilt servos, Octoprint and my babycam projects would be complete.

jacksonliam, 2017-08-12:
@devekrehbiel that's unrelated to what's being discussed here, needs its own issue. But basically the answer is no. 
You could do GPU accelerated yuv to jpeg like we do in the Pi cam plugin but it's not a simple thing to add and the code would be hardware specific. The CPU load is resolution and framerate specific so drop down to 15fps and use half the CPU time ;-)

DrLous, 2018-03-23:
Hi,

Need modification : 
**In plugins/input_uvc/CMakeLists.txt
add new line "find_library(JPEG_LIB jpeg)" after the line "find_library(V4L2_LIB v4l2)"**

And the -y work :)

But for YUV camera (in my case usbtv door camera), the motion software without motion part use less cpu for streaming.