jacksonliam/mjpg-streamer #64: ISO setting ignore?

totalitarian, 2017-01-22:
Is the ISO setting implemented?

jacksonliam, 2017-01-22:
For uvc or raspicam?

totalitarian, 2017-01-22:
raspicam 

-fps 30 -x 640 -y 480 -ISO 100 -vf -vs

totalitarian, 2017-01-22:
This page also confirms they are having issues with ISO

jacksonliam, 2017-01-22:
The code looks like it should work but I can't say I've tried it. 
Are there any error messages? Could you check the exif?  Have you tried a few different settings?

totalitarian, 2017-01-22:
Sorry this page http://www.ostafichuk.com/raspberry-pi-projects/using-a-server-for-remote-motion-detection-from-raspberry-pi-or-any-device-that-can-do-mjpeg-streaming/

Not sure how to check the exif on an mpeg. It works with raspistill

totalitarian, 2017-01-22:
Checked the snapshot image and doesn't have the ISO value in the exif

jacksonliam, 2017-01-22:
Thanks for trying, can you try in stills mode (-usestills) with an fps of 5

totalitarian, 2017-01-22:
OK now I can see the exif, iso is set to 1000 despite me setting it to 100

Serveurperso, 2017-09-18:
Anyway to set iso (more gain) at high frame rate ? raspicam is useless for indoor realtime robotics without this !

Serveurperso, 2017-09-18:
exposure compensation -ev do not work also:(