jacksonliam/mjpg-streamer #65: input_raspicam argument -awb off results in black image

derbroti, 2017-01-23:
disabling awb results is a completely black image 

- when using no other options 
- even with awbgainR/B set to any possible value the image stays clean black - no noise or anything

is this a bug here or is the hardware weird?

jacksonliam, 2017-01-23:
Assuming it works in other apps? And you get a fine image with awb enabled? 

Which raspbian versions and firmware version?

derbroti, 2017-01-23:
sorry for the brief report:

using for example: raspistill -awb off -awbg 5.5,5.5 creates a nice violet tinted picture.

using awb <anything else> i get a nice picture with mjpg-streamer

using: 
raspicam v1 - rev1.3
OS:
raspbian jessie - the official release from: 2017-01-11
firmware: 
Nov 25 2016 16:09:34 
Copyright (c) 2012 Broadcom
version 48a26a2ae46c497139b3d5a9c8d15485c7b3bfbc (clean) (release)

derbroti, 2017-01-26:
I am sorry, after comparing the raspistill code and yours, I finally found the problem -...-

I was calling the streamer with: "./mjpg_streamer -i "./input_raspicam.so -fps 10 -awb off -awbgainR 5.0 awbgainB 5.0 [...]"

See the problem? -> I forgot the dash for the second gain parameter... and that results in no warning or any info but a plain black video -> when I re-add the dash it works perfectly...

Sorry if you spend time searching for that bug - but is it possible to enforce the presents of the gain parameter? or at least show the values in the listing, where all other parameters are shown when staring the streamer - I would probably would have noticed the missing second gain then

jacksonliam, 2017-01-27:
Interesting, glad it's working for you now! 

I would certainly expect missing the dash to result in an error! I'll see what I can do.

jacksonliam, 2017-01-27:
Created #66 to deal with arguments issue

closed 2017-01-27 by jacksonliam