jacksonliam/mjpg-streamer #69: input_opencv, what's version of python and numpy using?

jadorre, 2017-02-17:
Hello,

You said input_opencv has been tested with OpenCV 3.1.0 but what version of python and numpy do you use? I installed OpenCV 3.1.0 with Python 3.4.2 and numpy 1.12.0 and i've got errors with these versions.
Thanks

ps: sorry for my poor English :(

virtuald, 2017-02-17:
Those should work. What errors?

jadorre, 2017-02-17:
i tried many thing. First i install opencv through this pre-compiled binary and the error was importerror, multiarray, undefined symbol PyExc_SystemError. https://github.com/jabelone/OpenCV-for-Pi. So i tried to compile the binary with this tuturial http://www.pyimagesearch.com/2016/04/18/install-guide-raspberry-pi-3-raspbian-jessie-opencv-3/ and the error was multiarray...undefined symbol "pytype_genericnew". And now i try to compile opencv with python 2.7...

jadorre, 2017-02-18:
With Python 2.7.9, just can't build input_opencv plugin. Same error as here : http://pastebin.com/M6Wnkcnw

No way.

virtuald, 2017-02-19:
I've heard of others having similar issues, and I don't really have a good answer for you. Something weird changed in the last year, and I don't personally have the time to go debug it.

Instead, if you're looking for a package that can capture images from USB devices and stream out via HTTP (among many other things), I recommend using [robotpy-cscore](https://github.com/robotpy/robotpy-cscore), as it's significantly more flexible than mjpg-streamer and this plugin.

jadorre, 2017-02-19:
Thanks for the recommendation. I took a look at the robotpy-score code and the call to numpy looks identical which doesn't suggest anything good...

Concerning python 2.7.9, this version doesn't support the new api PyUnicode_DecodeFSDefault and that's why mjpg-streamer doesn't compile.

However, i don't understand why my installation does not work. Here's the output of the compilation :

pi@raspberrypi:/usr/src/mjpg-streamer/mjpg-streamer-experimental $ make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
-- Found OpenCV: /usr/local (found version "3.2.0") found components:  core imgproc highgui videoio 
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython3.4m.so (found version "3.4.2") 
-- Found PythonInterp: /usr/bin/python3.4 (found version "3.4.2") 
-- Found NUMPY: /usr/local/lib/python3.4/dist-packages/numpy/core/include  
-- Found components for NumPy
-- NUMPY_ROOT_DIR    = /usr/local
-- NUMPY_INCLUDES    = /usr/local/lib/python3.4/dist-packages/numpy/core/include
-- NUMPY_LIBRARIES   = 
-- NUMPY_API_VERSION = 1.12.0
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- Checking for module 'libgphoto2'
--   No package 'libgphoto2' found
-- Could NOT find GPHOTO2 (missing:  GPHOTO2_LIBRARY GPHOTO2_INCLUDE_DIR) 
-- Looking for include file linux/videodev2.h
-- Looking for include file linux/videodev2.h - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Could NOT find SDL (missing:  SDL_LIBRARY SDL_INCLUDE_DIR) 
-- 
-- The following features have been enabled:

 * PLUGIN_INPUT_FILE , File input plugin
 * PLUGIN_INPUT_HTTP , HTTP input proxy plugin
 * PLUGIN_INPUT_OPENCV , OpenCV input plugin
 * PLUGIN_CVFILTER_CPP , OpenCV example filter
 * PLUGIN_CVFILTER_PY , OpenCV python filter
 * PLUGIN_INPUT_RASPICAM , Raspberry Pi input camera plugin
 * PLUGIN_INPUT_UVC , Video 4 Linux input plugin
 * PLUGIN_OUTPUT_FILE , File output plugin
 * PLUGIN_OUTPUT_HTTP , HTTP server output plugin
 * PLUGIN_OUTPUT_RTSP , RTSP output plugin
 * PLUGIN_OUTPUT_UDP , UDP output stream plugin

-- The following OPTIONAL packages have been found:

 * OpenCV
 * PythonLibs
 * PythonInterp
[... 236 lines cut ...]
--     ant:                         NO
--     JNI:                         NO
--     Java wrappers:               NO
--     Java tests:                  NO
-- 
--   Matlab:                        Matlab not found or implicitly disabled
-- 
--   Documentation:
--     Doxygen:                     NO
-- 
--   Tests and samples:
--     Tests:                       YES
--     Performance tests:           YES
--     C/C++ Examples:              NO
-- 
--   Install path:                  /usr/local
-- 
--   cvconfig.h is in:              /home/pi/opencv-3.2.0/build
-- -----------------------------------------------------------------
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/opencv-3.2.0/build

What's wrong ?

virtuald, 2017-02-19:
mjpg-streamer's py plugin is trying to embed python inside of it, and there's *something* wrong about the way I went about it. Thus why it compiles, but you get errors at runtime. Figure out what that something is, and it should work.

However, cscore is a python module, so even though the numpy code is mostly the same, it's more likely to succeed because it is being used *from* python, so it already does the necessary things.

closed 2017-03-27 by jacksonliam