jacksonliam/mjpg-streamer #71: Option about FPS doesn't work

npasparuhov, 2017-03-05:
Hi guys, 

I'm glad to use this project, but we have a little problem with the fps option, it doesn't recognize it like it should be. I'm using regular camera module with max 30 fps on 1280x720 resolution.

With the command:
`sudo mjpg_streamer -i "/usr/local/lib/input_uvc.so -d /dev/video0 -n -r 1280x720 -f 15" -o "/usr/local/lib/output_http.so -n -w /usr/local/www -p 8080"`
It stream with 30fps.

Even with command:
`sudo mjpg_streamer -i "/usr/local/lib/input_uvc.so -d /dev/video0 -n -r 1280x720 -f 5" -o "/usr/local/lib/output_http.so -n -w /usr/local/www -p 8080"`
It stream with 30fps.

But the output is:
`i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 15/5 depends on the -f parameter
 i: Format............: MJPEG
 o: www-folder-path...: /usr/local/www/
 o: HTTP TCP port.....: 8080
 o: username:password.: disabled
 o: commands..........: disabled`

It always stream with max fps.
I have all packages it needs: 
subversion-tools libcairo2-dev libjpeg62-turbo-dev libpango1.0-dev libgif-dev build-essential g++ imagemagick v4l-utils

Also interesting is that when i put -quality parameter, the stream drop down to 5 fps. No matter what quality i put, it just stream with like 5fps.

We are using OrangePi Lite and the processor is quad core on 1.2 GHz.

jacksonliam, 2017-03-05:
Thanks, for the detailed report!
What is the camera? 

What is the output of `v4l2-ctl --list-formats-ext`

npasparuhov, 2017-03-06:
VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : MJPEG
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUV 4:2:2 (YUYV)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)

npasparuhov, 2017-03-06:
>  VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : MJPEG
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUV 4:2:2 (YUYV)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)

npasparuhov, 2017-03-06:
Index       : 0
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : MJPEG
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUV 4:2:2 (YUYV)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)

jacksonliam, 2017-03-06:
Can you see the problem? You tried 5 and 15 fps but your camera driver only supports 10 (in some yuv modes) or 30 fps. 

If what your request doesn't match what your camera supports, we run the camera at whatever it suggests we use (will be 30 in this case) and do software frame skipping.

How are you measuring the FPS of the stream?

npasparuhov, 2017-03-06:
VLC or Streaming engine shows me the fps.

Sorry for the misstake, it's my bad, can i change the source code somehow to put a maximum fps of 12-15 may be bacuse it generate so huge trafick and the wifi dongle go in so high temperature. I try on lower resolution and the dongle was fine, but i need that 1280x720 resolution.

Other option is to set up in low quality, but when i put --quality parameter the fps drop down to 4-5, which will not be fine for our project.

jacksonliam, 2017-03-06:
So the software frame skipping doesn't appear to be working. 

It relies on the timestamp from the camera, which may not be working, can you check the http headers (you can use the snapshot URL and [network inspection in ff ](http://stackoverflow.com/questions/33974595/in-firefox-how-do-i-see-http-request-headers-where-in-web-console)or chrome) to see if the "X-Timestamp"  header is correct for each frame?

It should be going up each time you refresh the page by the number of seconds you waited to refresh. If it is not changing or is 0 then we have found the problem.

closed 2017-03-27 by jacksonliam

pauledd, 2018-02-11:
I have the same issue. My camera is a:
ELP-USBFHD01M-BFV-D
```
ID 05a3:9230 ARC International
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'MJPG' (compressed)
        Name        : Motion-JPEG
                Size: Discrete 1920x1080
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.017s (60.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.008s (120.101 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.017s (60.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.008s (120.101 fps)

        Index       : 1
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2
                Size: Discrete 1920x1080
                        Interval: Discrete 0.167s (6.000 fps)
                Size: Discrete 1280x720
                        Interval: Discrete 0.111s (9.000 fps)
                Size: Discrete 1024x768
                        Interval: Discrete 0.167s (6.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 800x600
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.167s (6.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)

```
This cam does by default 60fps at 800x600 (mjpeg). I would like to lower that to something about 
15fps...
I start with:
```
mjpg_streamer -i "input_uvc.so -r 800x600 -ex 192 -f 15" -o output_http.so
MJPG Streamer Version: git rev: 8cc9d22c1e79905d529a248ccf05bbf0625e0bf3
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 800 x 600
 i: Frames Per Second.: 15
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 15 to 60
 i: Frame period time ......: 66 ms
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
 i: exposure mode     : 1
 i: absolute exposure : 192
^Csetting signal to stop
 i: cleaning up resources allocated by input thread
force cancellation of threads and cleanup resources
 o: cleaning up resources allocated by server thread #00
^Csetting signal to stop
force cancellation of threads and cleanup resources
Beendet

```
then I go to
IP:8080/?action=stream
but it shows nothing and loads forever, cannot CTRL+C, need to do killall mjpg_streamer.

koba1995, 2020-04-13:
I have similar issue. I found a part of the source code that could be causing this. 

### My situation
My camera is Buffalo BSWHD06M.
I specify the option -f 5 but actual fps was much higher.

`./mjpg_streamer -i "./input_uvc.so -d /dev/ video0 -n -r 640x480 -q 80 -f 5"  -o " ./output_http.so -w ./www -p 8080" `
> MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
>  i: Using V4L2 device.: /dev/video0
>  i: Desired Resolution: 640 x 480
>  i: Frames Per Second.: 5
>  i: Format............: JPEG
>  i: TV-Norm...........: DEFAULT
>  i: FPS coerced ......: from 5 to 16
>  o: www-folder-path......: ./www/
>  o: HTTP TCP port........: 8080
>  o: HTTP Listen Address..: (null)
>  o: username:password....: disabled
>  o: commands.............: enabled
> 

To check http header "":
> $ curl 127.0.0.1:8080/?action=stream 2>/dev/null | grep -a X-Timestamp
> X-Timestamp: 2223215.264295
> X-Timestamp: 2223215.396299
> X-Timestamp: 2223215.528302
> X-Timestamp: 2223215.664306
> X-Timestamp: 2223215.796309
> X-Timestamp: 2223215.932314
> X-Timestamp: 2223216.064316
> X-Timestamp: 2223216.196321
> X-Timestamp: 2223216.332324
> X-Timestamp: 2223216.464327
> X-Timestamp: 2223216.596331
> X-Timestamp: 2223216.732335
> X-Timestamp: 2223216.864339
> X-Timestamp: 2223216.996342
> X-Timestamp: 2223217.132348
> X-Timestamp: 2223217.264351
> X-Timestamp: 2223217.400357
> X-Timestamp: 2223217.532359
> X-Timestamp: 2223217.664362
> X-Timestamp: 2223217.800366
> ^C

Timestamp seems to be working correctly.

For reference `v4l2-ctl --list-formats-ext` command output:
> ioctl: VIDIOC_ENUM_FMT
>         Type: Video Capture
> 
>         [0]: 'YUYV' (YUYV 4:2:2)
>                 Size: Discrete 1280x720
>                         Interval: Discrete 0.125s (8.000 fps)
>                 Size: Discrete 800x600
>                         Interval: Discrete 0.067s (15.000 fps)
>                         Interval: Discrete 0.125s (8.000 fps)
>                 Size: Discrete 640x480
>                         Interval: Discrete 0.033s (30.000 fps)
>                         Interval: Discrete 0.050s (20.000 fps)
>                         Interval: Discrete 0.067s (15.000 fps)
>                         Interval: Discrete 0.125s (8.000 fps)
>                 Size: Discrete 640x360
>                         Interval: Discrete 0.033s (30.000 fps)
>                         Interval: Discrete 0.050s (20.000 fps)
>                         Interval: Discrete 0.067s (15.000 fps)
>                         Interval: Discrete 0.125s (8.000 fps)
>                 Size: Discrete 352x288
>                         Interval: Discrete 0.033s (30.000 fps)
>                         Interval: Discrete 0.050s (20.000 fps)
[... 79 lines cut ...]
>  i: Frame period time ......: 200 ms
>  o: www-folder-path......: ./www/
>  o: HTTP TCP port........: 8080
>  o: HTTP Listen Address..: (null)
>  o: username:password....: disabled
>  o: commands.............: enabled

"X-Timestamp" header: 
> $ curl 127.0.0.1:8080/?action=stream 2>/dev/null | grep -a X-Timestamp
> X-Timestamp: 2221811.890728
> X-Timestamp: 2221812.154738
> X-Timestamp: 2221812.422744
> X-Timestamp: 2221812.690751
> X-Timestamp: 2221812.958759
> X-Timestamp: 2221813.222768
> X-Timestamp: 2221813.490785
> X-Timestamp: 2221813.758793
> X-Timestamp: 2221814.022792
> X-Timestamp: 2221814.290798
> X-Timestamp: 2221814.558804
> X-Timestamp: 2221814.826813
> X-Timestamp: 2221815.090822
> ^C

Actual fps is not accurate but it has been near to specified value.

jacksonliam, 2020-04-13:
> I have similar issue. I found a part of the source code that could be causing this.
> 
> ### My situation
> My camera is Buffalo BSWHD06M.
> I specify the option -f 5 but actual fps was much higher.
> 
> `./mjpg_streamer -i "./input_uvc.so -d /dev/ video0 -n -r 640x480 -q 80 -f 5" -o " ./output_http.so -w ./www -p 8080" `
> 
> > MJPG Streamer Version: git rev: [501f636](https://github.com/jacksonliam/mjpg-streamer/commit/501f6362c5afddcfb41055f97ae484252c85c912)
> > i: Using V4L2 device.: /dev/video0
> > i: Desired Resolution: 640 x 480
> > i: Frames Per Second.: 5
> > i: Format............: JPEG
> > i: TV-Norm...........: DEFAULT
> > i: FPS coerced ......: from 5 to 16
> > o: www-folder-path......: ./www/
> > o: HTTP TCP port........: 8080
> > o: HTTP Listen Address..: (null)
> > o: username:password....: disabled
> > o: commands.............: enabled
> 
> To check http header "X-Timestamp":
> 
> > $ curl 127.0.0.1:8080/?action=stream 2>/dev/null | grep -a X-Timestamp
> > X-Timestamp: 2223215.264295
> > X-Timestamp: 2223215.396299
> > X-Timestamp: 2223215.528302
> > X-Timestamp: 2223215.664306
> > X-Timestamp: 2223215.796309
> > X-Timestamp: 2223215.932314
> > X-Timestamp: 2223216.064316
> > X-Timestamp: 2223216.196321
> > X-Timestamp: 2223216.332324
> > X-Timestamp: 2223216.464327
> > X-Timestamp: 2223216.596331
> > X-Timestamp: 2223216.732335
> > X-Timestamp: 2223216.864339
> > X-Timestamp: 2223216.996342
> > X-Timestamp: 2223217.132348
> > X-Timestamp: 2223217.264351
> > X-Timestamp: 2223217.400357
> > X-Timestamp: 2223217.532359
> > X-Timestamp: 2223217.664362
> > X-Timestamp: 2223217.800366
> > ^C
> 
> Timestamp seems to be working correctly.
> 
> For reference `v4l2-ctl --list-formats-ext` command output:
> 
> > ioctl: VIDIOC_ENUM_FMT
> > Type: Video Capture
> > ```
> >     [0]: 'YUYV' (YUYV 4:2:2)
> >             Size: Discrete 1280x720
> >                     Interval: Discrete 0.125s (8.000 fps)
> >             Size: Discrete 800x600
> >                     Interval: Discrete 0.067s (15.000 fps)
> >                     Interval: Discrete 0.125s (8.000 fps)
> >             Size: Discrete 640x480
> >                     Interval: Discrete 0.033s (30.000 fps)
> >                     Interval: Discrete 0.050s (20.000 fps)
> >                     Interval: Discrete 0.067s (15.000 fps)
> >                     Interval: Discrete 0.125s (8.000 fps)
> >             Size: Discrete 640x360
> >                     Interval: Discrete 0.033s (30.000 fps)
> >                     Interval: Discrete 0.050s (20.000 fps)
> >                     Interval: Discrete 0.067s (15.000 fps)
> >                     Interval: Discrete 0.125s (8.000 fps)
> >             Size: Discrete 352x288
[... 94 lines cut ...]
> > X-Timestamp: 2221812.154738
> > X-Timestamp: 2221812.422744
> > X-Timestamp: 2221812.690751
> > X-Timestamp: 2221812.958759
> > X-Timestamp: 2221813.222768
> > X-Timestamp: 2221813.490785
> > X-Timestamp: 2221813.758793
> > X-Timestamp: 2221814.022792
> > X-Timestamp: 2221814.290798
> > X-Timestamp: 2221814.558804
> > X-Timestamp: 2221814.826813
> > X-Timestamp: 2221815.090822
> > ^C
> 
> Actual fps is not accurate but it has been near to specified value.

That's not quite right,

On this version of mjpg-streamer we have disabled automatic software framedropping. 

Many camera drivers are poorly implemented and it was causing A LOT of issues. it also kept causing people confusion when their streams were skipping or latency was all over the place.

There was also no option for the user to chose which FPS to run their camera at if they wanted to use software framedropping, it just always tried to use the highest.

What we have is the `-softfps` option... So you can specify the FPS of the stream via framedropping, but you can still use the normal `-fps` option to select the FPS your camera should run at!