jacksonliam/mjpg-streamer #73: how to resize the output resolution ?

szhansen, 2017-03-27:
hi,I use bellow cmd  to generate 1080P video and watch it by : http://ip:8090?action=stream
but the Mjpeg 1920x1080 format video have huge data traffic,even i use local wifi,sometimes the video 
is  not smooth,but i can't reduce the input resolution,because i need take 1920x1080 photos by 
"http://ip:8090?action=snapshot",
so how could i take 1920x1080 photos but view the video by a low resoltion ?
-------------------------------------------------------------------------------------------------
`mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_raspicam.so -x 1920 -y 1080 -fps 25' -o '/usr/local/lib/mjpg-streamer/output_http.so -p 8090'`

vijayg78, 2017-03-27:
I have the same requirement, need the streaming at lower resolution but need to take snapshot at a high resolution.

jacksonliam, 2017-03-27:
I think it might be possible with the pi cam, and something I could see being useful. But not sure I can do it in a generic enough way that it'll be possible with usb cameras. 

I'll leave this issue here for now.

vijayg78, 2017-03-28:
One option will be to pass the resolution as part of parameter to URL (http://ipaddress:port/?action=stream&resolution=640x480).

szhansen, 2017-03-28:
@vijayg78 I have try your method,it's not work? could you give me detail ?
thank you so much.

vijayg78, 2017-03-28:
Oh no, this was a proposal to jacksonliam so that we can pass the resolution as part of URL. The question was how to generically provide the feature so the was just a suggestion. This will not work now.

szhansen, 2017-03-28:
@vijayg78  that's a good idea~

@jacksonliam will you fix this ?

thank you ~

filipgoc, 2017-04-02:
I would LOVE this.

jacksonliam, 2017-04-03:
@vijayg78 I don't think I can easily do it like that because of how the pi cam libraries are setup. The resolution for stills would probably have to be a command line parameter. 

I mean I need to find a way to implement it in the code generically so the other plugins can use it, but not breaking the ones that can't. USB cams probably can't do it without stopping the video to take the snapshot.

SirIcy, 2017-04-18:
This would be a great feature for PI cam
Will you implement it?

szhansen, 2017-06-07:
@jacksonliam ^_^

OutsourcedGuru, 2017-06-15:
In an implentation like OctoPrint (with a webcam running a 3D printer with a Raspi computer inside) then often the microSD is only 4GB.  Adafruit has recently discontinued the v1 version of the Pi NoIR camera and the new one captures/pushes a lot more resolution by default and it's possible that this is the reason why the video streaming feature is now intermittent for this setup.

So yes, it would be great to be able to back off the default resolution and to steer this programmatically.

CarlosGS, 2018-07-22:
Hi! Please check out #149 with my solution to serve a stream at multiple resolutions. It isn't pretty but it works a treat :-)

gdombiak, 2018-10-23:
Agreed with @OutsourcedGuru. Also using OctoPrint here. It would be so good to be able to pass a query param to indicate the desired resolution and solve it on the server. I'm now coding an Apple Watch app that will render the RPi camera (and other USB cameras). For the iOS app it is perfectly fine to use a higher resolution. However for the Apple Watch just to download such a big resolution takes like a second per frame event at 640 x 480. :( 

Any plans for implementing this?

Thanks,
Gaston

gpupo, 2020-10-23:
Any plans for implementing this?