jacksonliam/mjpg-streamer #74: how to record a video file?

szhansen, 2017-03-27:
hi, anybody have a resoltion to record the video use mjpg-streamer ?
I try to use ffmpeg,but the video frame is quit low,performance is bad.

ffmpeg -i http://ip:8090/?action=stream -qscale 2 -vcodec mpeg4 out.mov

jacksonliam, 2017-03-27:
There is an output-file.so plugin that can store an mjpeg file             
```
" The following parameters can be passed to this plugin:\n\n" \
            " [-f | --folder ]........: folder to save pictures\n" \
            " [-m | --mjpeg ].........: save the frames to an mjpg file \n" \
            " [-d | --delay ].........: delay after saving pictures in ms\n" \
            " [-i | --input ].........: read frames from the specified input plugin\n" \
            " The following arguments are takes effect only if the current mode is not MJPG\n" \
            " [-s | --size ]..........: size of ring buffer (max number of pictures to hold)\n" \
            " [-e | --exceed ]........: allow ringbuffer to exceed limit by this amount\n" \
            " [-c | --command ].......: execute command after saving picture\n"\
            " ---------------------------------------------------------------\n");
```

closed 2017-03-27 by jacksonliam

szhansen, 2017-03-28:
@jacksonliam Unfortunately，it's not work.
I use:
`mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_raspicam.so -x 1920 -y 1080 -fps 25 -quality 10' -o '/usr/local/lib/mjpg-streamer/output_http.so -p 8090' -o '/usr/local/lib/mjpg-streamer/output_file.so -m test.mjpeg'
`
I got a file: /tmp/test.mjpeg, about 100Mb,and copy it to windows ,play it with vlc,but it's only show a frame picture,not a continue video picture.

I also test the bellow cmd,have the same results:
`mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_raspicam.so -x 1920 -y 1080 -fps 25 -quality 10' -o '/usr/local/lib/mjpg-streamer/output_file.so -m test.mjpeg'
`
`mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_uvc.so -d /dev/video0 -r 1920x1080 -f 25' -o '/usr/local/lib/mjpg-streamer/output_file.so -m test.mjpeg'
`

jacksonliam, 2017-03-28:
I don't think VLC will play a raw mjpeg file. Try putting it in a container or encoding it to a video file. 

Something like this might work (not tested it) 
`ffmpeg - framerate 25 -i test.mjpeg -vcodec copy out.avi`

szhansen, 2017-03-29:
@jacksonliam Great,It's work .
Thank you so much.

vamsitungala, 2018-02-09:
I am getting just pictures in /tmp/ folder no .mjpeg file

vamsitungala, 2018-02-10:
i want to save .mjpg file with date and time. how is it possible.

jacksonliam, 2018-02-10:
@vamsitungala use a lower resolution or framerate (or both). Or you could re-encode the video with ffmpeg  after and use a better codec (e.g h264).

vamsitungala, 2018-02-12:
Is it possible to convert while recording?