jacksonliam/mjpg-streamer #75: Issue with input_http.so: Can't allocate socket

guanche, 2017-03-27:
I'm trying to fetch a jpeg URL using the http input plugin using: ```mjpg_streamer -i "input_http.so -H http://www.example.com/image.jpg -p 80" -o "output_http.so"```

Which produces te following error:

```
MJPG Streamer Version.: 2.0
 i: host.............: www.webcamcenter.nl/flower.jpg
 i: port.............: 80
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
Name or service not known: Success
Can't allocate socket, will continue probing
: Invalid argument
Can't allocate socket, will continue probing
: Invalid argument
Segmentation fault
```

When using the file input and wget'ing the file everything works as fine, so the module is working. But I really don't want to destroy my SD cards using this method 24/7 :)

Many thanks in advance!

jacksonliam, 2017-03-27:
The host should just be an address, not a URI. (not sure if FQDN works, might need to be an IP). 

The request is always GET /?action=stream HTTP/1.0
This plugin is really only made for talking to other mjpg-streamer instances. If you want to do a PR to improve it into a more generic client feel free :-) but it's not something I'll be taking on! 

If you want to do the wget thing, use a ramdisk to avoid thrashing storage.

closed 2017-03-27 by jacksonliam

guanche, 2017-03-27:
I figured something like that after looking a bit better through the code. My C isn't proficient enough though. But thanks for the suggestion on using the ramdisk!