jacksonliam/mjpg-streamer #76: how to use input_http to connect a  localhost mjpeg stream?

szhansen, 2017-03-30:
I input the following cmd in the same Pi3,but get a error:
Firstly:
`mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_raspicam.so -x 1920 -y 1080 -fps 25 -quality 10' -o '/usr/local/lib/mjpg-streamer/output_http.so -p 8090'`
And then:
`mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_http.so -p 8090' -o '/usr/local/lib/mjpg-streamer/output_file.so -f /var/www/videos/ -m inputhttp.mjpeg'`

Got the following log:
`root@raspberrypi:/home/pi# mjpg_streamer -i '/usr/local/lib/mjpg-streamer/input_http.so -p 8090' -o '/usr/local/lib/mjpg-streamer/ou
tput_file.so -f /var/www/videos/ -m inputhttp.mjpeg'                                                                        
MJPG Streamer Version.: 2.0                                                                                                         
 i: host.............: localhost                                                                                                    
 i: port.............: 8090                                                                                                         
 o: output folder.....: /var/www/videos                                                                                     
 o: input plugin.....: 0: /usr/local/lib/mjpg-streamer/input_http.so                                                                
 o: delay after save..: 0                                                                                                           
 o: output file.......: /var/www/videos/inputhttp.mjpeg                                                                     
Segmentation fault                                                                                                                  
root@raspberrypi:/home/pi#      `

'Segmentation fault'
can't i input two mjpeg_streamer cmd in the same pi ?
but i actually need save video to a file when output mjpeg to http at the same time.

@jacksonliam could you give me some guides? thank you so much.

jacksonliam, 2017-03-30:
You should be able to run two with different cameras. Though I think you need to provide a host to the input_http plugin, which might be why it's failing. (but we shouldn't be segfaulting, I'll look at that). 

Anyway, I don't think you need to run two, you should be able to specify two output plugins. Just put another -o after the first.

szhansen, 2017-03-30:
@jacksonliam only can be one camera for my  application ,i design it look like a ip camera,
user can watch the video by :http://ip:8090/?action=stream,
and also provide user a button like 'record' to start or stop record the video to a mjpeg file at any time ,
so , that's my application, please help
thank you so much.

jacksonliam, 2017-08-28:
I think I've fixed the segfault in input_http, but the frames need to be below 256k, memory ideally would be dynamically allocated based on incoming frame size, but not got time to do that right now.

commit c1ecfaf7 (2017-08-28): Fix memory / segfault issue in input_http
Fixes #76
closed 2017-08-28 by jacksonliam