jacksonliam/mjpg-streamer #79: domoticz

Blacksly, 2017-04-14:
Hi guys ! 

I have installed mjpg-streamer in my pi zero w with a camera pi and on another pi i have installed domoticz. 
Do you know how i can catch the video for create camera in domoticz ?? 

Thank you ! 

(sorry for my english)

jacksonliam, 2017-04-14:
That looks OK (assuming you did start mjpg-streamer on port 9000).

Try one of these for the URL setting:
`/?action=snapshot`
`?action=snapshot`

Blacksly, 2017-04-14:
Hey, thanks for your help ! 
Yes, mjpg is strating on port 9000
and i had try /?action=snapshot and ?action=snapshot, dosen't work too :/

jacksonliam, 2017-04-14:
OK, and you can see the snapshot if you go to 192.168.1.17:9000/?action=snapshot in your browser?

Blacksly, 2017-04-14:
Yep, I have the snapshot ^_^

jacksonliam, 2017-04-14:
Then unfortunately I don't think I can help, sounds like mjpg-streamer is working fine, I'd go to domoticz for help. 
Perhaps they don't handle URLs without segments or with query parameters?

closed 2017-04-14 by jacksonliam

Blacksly, 2017-04-14:
Okay, thanks for your help, I will go on Domoticz forum and I will try to reinstall Domoticz ^_^ thanks!