jacksonliam/mjpg-streamer #84: output_http: is it possible to get the number of currently connected clients?

zpfvo, 2017-05-31:
I'd like to know how many clients are connected at a given moment.
I enabled the `MANAGEMENT`-feature and now i can get a json with the connected clients and a timestamp of the latest received frame.
````.json

{
  "clients": [
    {
      "address": "127.0.0.1",
      "timestamp": 1496240473
    }
  ]
}
`````

Is it correct that every client that connects gets an entry in this JSON that will never expire? And if i wanted to check if any of those clients is still connected i need to check, wether a timestamp is older than some seconds?

jacksonliam, 2017-05-31:
I can't say I've used this feature, but I imagine it's a list of currently streaming clients and the last time they got a frame. 

Let me know if you find anything out!

Out of interest which method did you use to enable the `MANAGEMENT` define?

closed 2017-05-31 by jacksonliam

zpfvo, 2017-06-01:
I compiled with CMake and used the line:
`cmake .. -DENABLE_HTTP_MANAGEMENT=ON`