jacksonliam/mjpg-streamer #85: an example to use output_rtsp please

romanten2017, 2017-07-08:
Http is so nice.But I need to use rtsp for my project. 
Can you help me with some suggestion?
Thank so much.

lzlcss, 2017-08-09:
also need an example, i use output_rtsp.so to run program, but cannot get stream in PC part via VLC

romanten2017, 2017-08-19:
So do I.

jacksonliam, 2017-08-28:
I think this is one of the more experimental plugins. I didn't write it and haven't used it. What do you want to use as the RTSP client? 

Here's the options, so it's just about specifying which port to listen on:
```
            " ---------------------------------------------------------------\n" \
            " The following parameters can be passed to this plugin:\n\n" \
            " [-p | --port ]..........: UDP port to listen for picture requests. UDP message is the filename to save\n\n" \
            " [-i | --input ].......: read frames from the specified input plugin (first input plugin between the arguments is the 0th)\n\n" \
            " ---------------------------------------------------------------\n");
```

closed 2017-08-28 by jacksonliam

ighack, 2018-03-31:
mjpg_streamer -i "input_uvc.so -y" -o "output_rtsp.so -p 8081"
and I via VLC rtsp://pi_of_ip:8081.
I can't see anything