jacksonliam/mjpg-streamer #87: Compilation error

VictorLamoine, 2017-08-13:


jacksonliam, 2017-08-13:
What commands did you use to build? I assume you have installed opencv and want to use the opencv plugin?

VictorLamoine, 2017-08-13:
I am using the installation instructions from [OctoPrint](https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#webcam).

I'm currently using these instructions on my desktop (so, I'm not on the RPi) to test some webcams :)

VictorLamoine, 2017-08-13:
My CMakeCache.txt:

```cmake
# This is the CMakeCache file.
# For build in directory: /home/victor/libraries/mjpg-streamer/mjpg-streamer-experimental/_build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: None(CMAKE_CXX_FLAGS or
// CMAKE_C_FLAGS used) Debug Release RelWithDebInfo MinSizeRel.
CMAKE_BUILD_TYPE:STRING=Release

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler
CMAKE_CXX_COMPILER:FILEPATH=/usr/lib/ccache/c++

//Flags used by the compiler during all build types.
CMAKE_CXX_FLAGS:STRING='-fdiagnostics-color  '

//Flags used by the compiler during debug builds.
CMAKE_CXX_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during release builds for minimum
// size.
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the compiler during release builds.
CMAKE_CXX_FLAGS_RELEASE:STRING=-O3 -DNDEBUG

//Flags used by the compiler during release builds with debug info.
CMAKE_CXX_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG

//C compiler
CMAKE_C_COMPILER:FILEPATH=/usr/lib/ccache/cc

//Flags used by the compiler during all build types.
CMAKE_C_FLAGS:STRING=

//Flags used by the compiler during debug builds.
CMAKE_C_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during release builds for minimum
// size.
CMAKE_C_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG

//Flags used by the compiler during release builds.
CMAKE_C_FLAGS_RELEASE:STRING=-O3 -DNDEBUG

//Flags used by the compiler during release builds with debug info.
CMAKE_C_FLAGS_RELWITHDEBINFO:STRING=-O2 -g -DNDEBUG

//Flags used by the linker.
CMAKE_EXE_LINKER_FLAGS:STRING=

[... 386 lines cut ...]
PC_GPHOTO2_STATIC_CFLAGS:INTERNAL=
PC_GPHOTO2_STATIC_CFLAGS_I:INTERNAL=
PC_GPHOTO2_STATIC_CFLAGS_OTHER:INTERNAL=
PC_GPHOTO2_STATIC_LIBDIR:INTERNAL=
PC_GPHOTO2_STATIC_LIBS:INTERNAL=
PC_GPHOTO2_STATIC_LIBS_L:INTERNAL=
PC_GPHOTO2_STATIC_LIBS_OTHER:INTERNAL=
PC_GPHOTO2_STATIC_LIBS_PATHS:INTERNAL=
PC_GPHOTO2_VERSION:INTERNAL=
PC_GPHOTO2_libgphoto2_INCLUDEDIR:INTERNAL=
PC_GPHOTO2_libgphoto2_LIBDIR:INTERNAL=
PC_GPHOTO2_libgphoto2_PREFIX:INTERNAL=
PC_GPHOTO2_libgphoto2_VERSION:INTERNAL=
//ADVANCED property for variable: PKG_CONFIG_EXECUTABLE
PKG_CONFIG_EXECUTABLE-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_EXECUTABLE
PYTHON_EXECUTABLE-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_INCLUDE_DIR
PYTHON_INCLUDE_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_LIBRARY
PYTHON_LIBRARY-ADVANCED:INTERNAL=1
//ADVANCED property for variable: PYTHON_LIBRARY_DEBUG
PYTHON_LIBRARY_DEBUG-ADVANCED:INTERNAL=1
__pkg_config_checked_PC_GPHOTO2:INTERNAL=1
```

VictorLamoine, 2017-08-13:
# OpenCV packages on my system

```bash
$ dpkg -l | grep opencv
ii  libopencv-calib3d-dev:amd64                                 2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-calib3d
ii  libopencv-calib3d2.4v5:amd64                                2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision Camera Calibration library
ii  libopencv-contrib-dev:amd64                                 2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-contrib
ii  libopencv-contrib2.4v5:amd64                                2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision contrib library
ii  libopencv-core-dev:amd64                                    2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-core
ii  libopencv-core2.4v5:amd64                                   2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision core library
ii  libopencv-dev                                               2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for opencv
ii  libopencv-features2d-dev:amd64                              2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-features2d
ii  libopencv-features2d2.4v5:amd64                             2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision Feature Detection and Descriptor Extraction library
ii  libopencv-flann-dev:amd64                                   2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-flann
ii  libopencv-flann2.4v5:amd64                                  2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision Clustering and Search in Multi-Dimensional spaces library
ii  libopencv-gpu-dev:amd64                                     2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-gpu2.4v5
ii  libopencv-gpu2.4v5:amd64                                    2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision GPU library
ii  libopencv-highgui-dev:amd64                                 2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-highgui
ii  libopencv-highgui2.4v5:amd64                                2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision High-level GUI and Media I/O library
ii  libopencv-imgproc-dev:amd64                                 2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-imgproc
ii  libopencv-imgproc2.4v5:amd64                                2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision Image Processing library
ii  libopencv-legacy-dev:amd64                                  2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-legacy
ii  libopencv-legacy2.4v5:amd64                                 2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision legacy library
ii  libopencv-ml-dev:amd64                                      2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-ml
ii  libopencv-ml2.4v5:amd64                                     2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision Machine Learning library
ii  libopencv-objdetect-dev:amd64                               2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-objdetect
ii  libopencv-objdetect2.4v5:amd64                              2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision Object Detection library
ii  libopencv-ocl-dev:amd64                                     2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-ocl2.4v5
ii  libopencv-ocl2.4v5:amd64                                    2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision OpenCL support library
ii  libopencv-photo-dev:amd64                                   2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-photo2.4v5
ii  libopencv-photo2.4v5:amd64                                  2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision computational photography library
ii  libopencv-stitching-dev:amd64                               2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-stitching2.4v5
ii  libopencv-stitching2.4v5:amd64                              2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision image stitching library
ii  libopencv-superres-dev:amd64                                2.4.9.1+dfsg-1.5ubuntu1                               amd64        development files for libopencv-superres2.4v5
ii  libopencv-superres2.4v5:amd64                               2.4.9.1+dfsg-1.5ubuntu1                               amd64        computer vision Super Resolution library
[... cut ...]
ii  python-opencv                                               2.4.9.1+dfsg-1.5ubuntu1                               amd64        Python bindings for the computer vision library
ii  ros-kinetic-opencv3                                         3.2.0-4xenial-20170607-095735-0800                    amd64        OpenCV 3.x
ii  ros-kinetic-vision-opencv                                   1.12.4-0xenial-20170613-183914-0800                   amd64        Packages for interfacing ROS with OpenCV, a library of programming functions for real time computer vision.
```

VictorLamoine, 2017-08-13:
Disabling `PLUGIN_CVFILTER_PY` allows to compile.

## Note to beginners
```bash
sudo apt install -yqq cmake-curses-gui
cd _build
ccmake .
```

Go to the `PLUGIN_CVFILTER_PY` line, set it to `OFF`. Press `C` to configure, then `G` to generate.
When done, call `make` again.

VictorLamoine, 2017-08-13:
Looks like it's failing because I have OpenCV 3.2.0 installed from the [ROS](ros.org) repository.
I did not try but I think using the system (OpenCV 2.8) would work.

I don't think I need this plug-in so it's fine disabling it in my case.

jacksonliam, 2017-08-28:
Thanks for looking into this further and posting a workaround! 

I think this is actually related to having multiple versions of python installed (2.7 and 3.x) and something is picking up the 2.7 
header rather than the 3.x

I'm not sure if there's anything I can do about this on my side except for disable the opencv plugin from building by default.

closed 2017-08-28 by jacksonliam

ayourk, 2020-05-24:
The fix is to have CMake disable the PY part of the plugin if Python 3 is not going to be the Python version to be used:
[opencv_requires_python3.patch.txt](https://github.com/jacksonliam/mjpg-streamer/files/4674374/opencv_requires_python3.patch.txt)

OR make the filter Python 2 compatible:
[opencv_python2_compat.patch.txt](https://github.com/jacksonliam/mjpg-streamer/files/4674396/opencv_python2_compat.patch.txt)

I'm not familiar with what it would take to fully convert these 2 routines to Python 2 compatibility, so I simply neutered them to not allow a script file to be passed.

jacksonliam, 2020-07-03:
> The fix is to have CMake disable the PY part of the plugin if Python 3 is not going to be the Python version to be used: [...]

Thanks, that seems like a sensible change.