jacksonliam/mjpg-streamer #88: Add -rot to input_uvc.so

b-morgan, 2017-08-28:
Please add -rot to input_uvc.so. Implementation can mirror the -rot implementation in input_raspi.so

jacksonliam, 2017-08-28:
The rotate and flip are supported by input_uvc plugin but may depend on the connected hardware/driver supporting it
```
[-rot ]................: Set image rotation (0-359)
[-hf ].................: Set horizontal flip (true/false)
```

closed 2017-08-28 by jacksonliam

dvv, 2017-08-29:
It can't be done in software by jtransform or manually, at least for the simplest cases such as 90 degree rotation?
TIA

jacksonliam, 2017-08-29:
Probably can be done with libjepg, to avoid adding a new dependency, but it will be more cpu intensive. Feel free to do a PR add software rotation as an extra command line option.