jacksonliam/mjpg-streamer #91: very dark indoor image at 25fps

Serveurperso, 2017-09-18:
The -iso parameter do not work, I tested -ev without success also.
How I can get the high gain (and noisy) video to make any indoor robotics use ?
I need >= 25FPS because lower framerate is useless for my project.

Serveurperso, 2017-09-18:
You can test https://www.serveurperso.com/?page=robot original CCD camera from my robotic project
Versus http://www.serveurperso.com:8080/ raspicam version very dark:(

I made this launcher script :
```
(root|~/scripts) cat mjpg-streamer.sh
#!/bin/bash

pat=/root/mjpg-streamer/mjpg-streamer-experimental
bin=$pat/mjpg_streamer

i="input_raspicam.so -fps 25 -q 10 -x 720 -y 576 -rot 180"
o="output_http.so -p 8080"

LD_LIBRARY_PATH=$pat $bin -i "$i" -o "$o" -b
```

the -ex night do nothing also:(

Serveurperso, 2017-09-18:
CCD Super HAD2 camera + complex analog video downlink :
[image]

Raspicam :
[image]

jacksonliam, 2017-09-18:
Have you tried raspivid directly? Are you able to get a decent image out of that?

Serveurperso, 2017-09-18:
No tested raspivid for now. Do you think the ISO param work with raspivid ?
I think it's the same limitation if you forward all parameter offered by raspivid ?

jacksonliam, 2017-09-18:
If it doesn't work in raspivid there's not a lot I can do, but if it does then it's probably something I can fix, so it would be useful if you could try raspivid (with the same resolution/framerate) and let me know how it goes!

Serveurperso, 2017-09-19:
hi, i tested

raspivid -fps 25 -w 720 -h 576 -ISO 800 -cd MJPEG -o /var/www/temp/video.mjpg

raspivid -fps 25 -w 720 -h 576 -mm average -ISO 800 -cd MJPEG -o /var/www/temp/video.mjpg

without success. always too dark...

Serveurperso, 2017-09-19:
I tested all combination of exposure night/auto/off :
-ISO,  -ev and -mm are ignored

Serveurperso, 2017-09-19:
I opened a issue on raspberry pi https://github.com/raspberrypi/userland/issues/424#issuecomment-330462594

Serveurperso, 2017-09-19:
when I add the -drc high (dynamic range compensation to maximum) parameter the mjpeg streamer launch but there is no video stream...

jacksonliam, 2017-09-19:
Looking at the other issue this appears to be a hardware problem so I'll close this for now, you might get more luck asking for tips on the raspberry pi forum on how to get the best low light video performance out of the sensor.

closed 2017-09-19 by jacksonliam