jacksonliam/mjpg-streamer #93: Unable to get my endoscope to work on the Pi 3

GrueMaster, 2017-10-08:
I bought [this endoscope](http://a.co/0cjpkFT) and I can get it to work on mjpg-streamer running Ubuntu 16.04 on my Intel NUC 6i5SYB, but not on a Pi 3 running octopi (tried building the latest git commit so they match).

cmdline:  ./mjpg_streamer -o 'output_http.so -n -w ./www/' -i 'input_uvc.so -y -q 75 -f 10'

It doesn't appear to be using much cpu power.  And my SD is only half-full (4G total, 2G available).  Tomorrow I will try switching to a 16G SD, but I don't think that will make much of a difference.

Anything you can suggest?

cms-, 2017-10-08:
Hi, what is the output of the program after you run it? What happens when you connect to port 8080 of your Pi3's IP address?

jacksonliam, 2017-10-08:
With the device plugged in can you print the output of `ls /dev/video*` on both pi and Ubuntu?

Also the output of running `lsusb` may be helpful in determining the type of device. 

I imagine it's the case of a missing driver on raspbian on the pi. You might try one of the Ubuntu distributions for the pi.

GrueMaster, 2017-10-08:
Nevermind, I figured it out.  Needed to remove the '-f 10' from the input plugin options.

Works with Octopi now.  :+1:

closed 2017-10-09 by jacksonliam