jacksonliam/mjpg-streamer #98: How to use the -cfx with input_raspicam.so?

LinuxCircle, 2017-11-10:
I tried mjpg_streamer -o "output_http.so -w /home/pi/Projects/mjpg-streamer/mjpg-streamer-experimental/www" -i "input_raspicam.so -x 320 -y 240 -fps 8 -rot 270 -cfx 1:10"

But resulting in:
MJPG Streamer Version.: 2.0
 i: fps.............: 8
 i: resolution........: 320 x 240
 i: camera parameters..............:

Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled Yes with U = 10, V = 128

It seems that the first parameter is totally ignored, and the second parameter was wrongly assigned as U variable.

jacksonliam, 2017-11-15:
Yep you're right there's a bug on https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/input_raspicam.c#L252

Will fix when I get the time ☺️

jacksonliam, 2018-01-25:
Fixed!

commit d3bf5f52 (2018-01-25): Fix Raspicam cfx read incorrectly. Fixes #98
closed 2018-01-25 by jacksonliam