jacksonliam/mjpg-streamer #100: MJPG-streamer stops streaming but no error found

frankforpresident, 2017-11-17:
I'm using MJPG-streamer via V4l2 with a capturing HDMI to USB module for testing purposes. In most cases this works perfect but when the source changes resolution or goes in standby, MJPG-streamer has 
difficulties continuing the stream (not always). When running the application in debug mode via "make CMAKE_BUILD_TYPE=Debug" no real cause was found.  

One thing that makes it extra strang is when opening the stream in VLC and in a browser, the stream in VLC continues but browser stops. The API's still respond but no data is found. New requests via "/?action=stream" of "/?action=snapshot" doesn't receive any data anymore

**Request with working API**
GET http://192.168.0.142:40000/?action=snapshot HTTP/1.1
User-Agent: Fiddler
Host: 192.168.0.142:40000

HTTP/1.0 200 OK
Connection: close
Server: MJPG-Streamer/0.2
Cache-Control: no-store, no-cache, must-revalidate, pre-check=0, post-check=0, max-age=0
Pragma: no-cache
Expires: Mon, 3 Jan 2000 12:34:56 GMT
Content-type: image/jpeg
X-Timestamp: 184659.604338

Format: JPEG
97,190 bytes

1,280w x 720h
0.11 bytes/px
96 dpi
Baseline
Subsample@4:2:2
APP0 Data (31 bytes)
Non-JFIF
HuffmanTables: 1

     !AVI1  x x                      C  
  

**Request with broken API (No data present)**
GET http://192.168.0.142:40000/?action=snapshot HTTP/1.1
User-Agent: Fiddler
Host: 192.168.0.142:40000

HTTP/1.0 200 OK
Connection: close
Server: MJPG-Streamer/0.2
Cache-Control: no-store, no-cache, must-revalidate, pre-check=0, post-check=0, max-age=0
Pragma: no-cache
Expires: Mon, 3 Jan 2000 12:34:56 GMT
Content-type: image/jpeg
X-Timestamp: 184701.364177

No body data

**Logging via CMAKE_BUILD_TYPE=Debug but without error**
` DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1574): create thread to handle client that just established a connection
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1577): serving client: 192.168.0.124
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1546): waiting for clients to connect
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, client_thread(), 1196): try to serve a file
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, client_thread(), 1233): parameter (len: 11): "stream.html"
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_file(), 728): stream.html EXTENSION: .html
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_file(), 746): trying to serve file "stream.html", extension: ".html" mime: "text/html"
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_file(), 758): opened file: ./www/stream.html
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, client_thread(), 1416): leaving HTTP client thread
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1574): create thread to handle client that just established a connection
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1577): serving client: 192.168.0.124
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, server_thread(), 1546): waiting for clients to connect
[... 112 lines cut ...]
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 531): sending boundary
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 508): got frame (size: 0 kB)
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 525): sending intemdiate header
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 528): sending frame
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 531): sending boundary
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c, cam_thread(), 692): copying frame from input: 0
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 508): got frame (size: 0 kB)
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 525): sending intemdiate header
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 528): sending frame
 DBG(/home/pi/mjpg-streamer-master/mjpg-streamer-experimental/plugins/output_http/httpd.c, send_stream(), 531): sending boundary`

More logging can be delivered on request.

**Command used to start the service**
`
export LD_LIBRARY_PATH=/usr/local/lib
mjpg_streamer -i "input_uvc.so -r 1280x720 -d /dev/video0" -o "output_http.so -p 8080 -w /usr/local/www"
`

The only way to escape this situation is to stop the service and rerun mjpg_streamer.

I getting the feeling "output_http.so" stops producing images and doesn't throw an error for some reason. Is there any way to increase the log level or to get this stable?

Thank you in advanced

frankforpresident, 2017-11-27:
Any chance of looking into this problem? It would be great if I could see when "output_http.so" is unable to sent data in it's body.

jacksonliam, 2017-11-27:
I expect the problem is with v4l2 and not the http plugin. The logs aren't useful in figuring out what's wrong. 

It may be that the plugin isn't handing a specific callback or something with the v4l2 api properly. Or it may be a bug with v4l2 or the device driver itself. 

Unfortunately I don't have any hardware which can change resolution (or the free time) to debug this myself. If you can figure out what's going on I might be able to point you in the right direction in the code to fix it (assuming it is an mjpg-streamer problem). 

Is there anything useful in the syslog when the res changes? 

Depending on your use case, it might be acceptable to setup a cron job which restarts mjpg-streamer if it stops serving frames, as a workaround.

frankforpresident, 2017-11-27:
Hi @jacksonliam ,

I reproduced the issue but sadly enough, no clear error message was thrown. Not in syslog, not in dmesg, not from mjpg-streamer.. The only way to know if the module and/or mjpg-streamer has crashed is by pulling a image and checking its body. 

Currenly 'curl http://localhost:8080?action=snapshot -o snapshot.jpg'  seems to only save the image when the streamer is 'active' and returns data.

I'm running mjpg-streamer in a docker container and I will write a healthcheck which will restart the container when no images are served via curl. Not ideal but should work as a workaround.

CapnBry, 2018-01-18:
I've run into what I think is the same issue as this, except using an ELP-USBFHD01M (ELP 1080P) USB camera. It works for a while but eventually will stop serving images. When pulling from the action=snapshot, the headers come but the request just hangs with no image data until the client times out. The server side thread is hung waiting for data from the input driver (backtrace is in output_http/client_thread()):
```
  14   Thread 0xb40ff460 (LWP 7732) "mjpg_streamer" 0xb6f9d7a4 in __pthread_cond_wait (
    cond=0x24708 <global+336>, mutex=0x246f0 <global+312>) at pthread_cond_wait.c:187
```
Looking into it I found that in the mjpg-streamer UVC handler, when there is a short frame received from the V4L2 layer, it does the [right thing and ignores it](https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c#L572).

However, the v4l2_buffer is not re-queued to the driver due to the code returning immediately-- it never gets down to the `VIDIOC_QBUF` ioctl, thus the input_uvc handler will eventually hang after it has dequeued all the buffers and not replenished them. The solution for me was to replace the `return 0;` with `break;`. The buffer is requeued, cam_thread() sees that the buffer size is 0 and loops again, the next image is bueno and **then everything just keeps working**.

Changing this one line of code has solved my camera hanging and requiring a restart of mjpg-streamer, at least for this deadlock condition, and I have been running several weeks now without any further issues.

jacksonliam, 2018-01-18:
Great debugging work @CapnBry, I'll look into a patch for this!

frankforpresident, 2018-01-18:
This would be awesome!

frankforpresident, 2018-01-19:
Hi @jacksonliam @CapnBry 

I've replaced 'return 0;' with 'break;' but this had no effect on my issue. But thank you for sharing! 

@jacksonliam, Would it be possible to let mjpg-streamer crash when v4l2 isn't providing any data after some amount of time? Just like when a device is detached? This would cause my docker container to auto-restart and restore the videolink.

CapnBry, 2018-01-19:
Dang @frankforpresident, that stinks, sorry it didn't help with your issue.

I should have also mentioned a symptom on my issue was that I'd see the message "Ignoring empty buffer ..." printed exactly 4 times every time mine failed, which lead me to the offending code. Now mine carries on after more than 4 times without issue.

Your situation is a bit unique in that it is dequeing buffers so it is able to respond to the HTTP request, but there's no actual data for the image. It would seem the the same code that was triggering in my case (data from the camera is smaller than a frame header) should detect the same condition in your case, but it isn't.  You might want to try changing the code [here in the http write](https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_http/httpd.c#L440) to add the content length of the frame data, so you can see if it is 0 at least. Something like this (which would be nice to have in the header anyway)
```
            "Content-Length: %d\r\n" \
            "X-Timestamp: %d.%06d\r\n" \
            "\r\n", frame_size, (int) timestamp.tv_sec, (int) timestamp.tv_usec);
```

frankforpresident, 2018-01-19:
Thanks @CapnBry, I will have a look into this.. Funny to see that they already included the content length [somewhere else...](https://github.com/jacksonliam/mjpg-streamer/blob/c1ecfaf7c5cb958cdfd251bdaa9824c3e798f890/mjpg-streamer-experimental/plugins/output_http/httpd.c#L522)

jacksonliam, 2018-01-25:
Thanks again @CapnBry - made that change (https://github.com/jacksonliam/mjpg-streamer/commit/740f3a0b7b26aa55fd0d63adce9a9a3260af89fa) 

@frankforpresident `git pull` and then `make` should get you running the latest code if you want to try it out.

jacksonliam, 2018-10-21:
@frankforpresident there's some new timeout stuff added in #164 which should do what you want.

closed 2018-10-21 by jacksonliam

frankforpresident, 2018-10-21:
@jacksonliam That would be awesome! Going to try it tomorrow.