jacksonliam/mjpg-streamer #101: Does not work with JPEG USB webcam

mijofa, 2017-12-06:
Using your version of mjpg-streamer-experimental with this V4L support USB webcam:
> 0ac8:301b Z-Star Microelectronics Corp. ZC0301 Webcam

Doesn't work, as it gives this error:

    $ ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -" -o "output_http.so -w ./www"
    MJPG Streamer Version.: 2.0
     i: Using V4L2 device.: /dev/video0
     i: Desired Resolution: 640 x 480
     i: Frames Per Second.: -1
     i: Format............: JPEG
     i: TV-Norm...........: DEFAULT
     i: Could not obtain the requested pixelformat: MJPG , driver gave us: JPEG
        ... will try to handle this by checking against supported formats.
     Init v4L2 failed !! exit fatal
     i: init_VideoIn failed

Normally I'd probably just go "oh well, this camera isn't supported" however it works absolutely fine on the same hardware when using codewithpassion's version of mjpg-streamer-experimental. For now I'm happy ot just use that one as I don't believe there's any newer features I'm wanting to use, I'm just a little unhappy that it hasn't been updated since 2012 and I'm relying on a dead project.

For context I've actually just set up [OctoPi](https://octopi.octoprint.org) which comes with this version of mjpg-streamer precompiled in for webcam feedback. I have however confirmed that this issue is still valid when compiling mjpg-streamer myself, but I don't know how well OctoPi will continue working with codewithpassion's version.

jacksonliam, 2017-12-06:
Interesting report, thanks.

Do you run the other version with the same command or do you use the -y flag?

Can you link me to the source of the other version and the specific webcam?

mijofa, 2017-12-06:
Used the exact same arguments, and neither version worked with the -y flag as the camer doesn't support YUVY at all.

I'm afraid I don't actually have any more info on the webcam than what I could pull from lsusb. I've had this sitting in my "random bits" box for a long time, and it's got no real reference to what brand/model it is or anything. [Here](https://user-images.githubusercontent.com/4810578/33657926-ecbfad0e-dace-11e7-8062-1ca3471640db.jpg) is an image I found on Google of what it looks like externally, but I doubt that's particularly useful since I've seen a few differently branded cameras with the same chassis.

ianespana, 2017-12-24:
I have the exact same issue on a VX-1000

betavr, 2017-12-27:
same here:

```
beta@sentry:~/src/mjpg-streamer/mjpg-streamer-experimental$ lsusb
Bus 008 Device 002: ID 0c45:6128 Microdia PC Camera (SN9C325 + OM6802)
```
```
beta@sentry:~/src/mjpg-streamer/mjpg-streamer-experimental$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'JPEG' (compressed)
        Name        : JFIF JPEG
                Size: Discrete 160x120
                Size: Discrete 320x240
                Size: Discrete 640x480
```
```
beta@sentry:~/src/mjpg-streamer/mjpg-streamer-experimental$ mjpg_streamer -i 'input_uvc.so' -o output_http.so
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: JPEG
    ... will try to handle this by checking against supported formats.
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

randUsr, 2018-01-05:
Same issue here. I used the original sources from sourceforge, and it worked.
`lsusb` gives  `Bus 001 Device 012: ID 0c45:614a Microdia `
Other commands give same results as @betavr.

It is a pretty old camera I had lying around.

Zotter, 2018-01-13:
In the hopes it's a help - I'm running into the same thing on Fedora 27. New, clean install on an HP laptop. Installed Octoprint and Setup mjpg_streamer using the instructions on https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian - section "Webcam".  

Yea, it's an ancient USB cam from the bucket-o-stuff.

$ lsusb
Bus 006 Device 002: ID 046d:08ad Logitech, Inc. QuickCam Communicate STX

$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'JPEG' (compressed)
	Name        : JFIF JPEG
		Size: Discrete 320x240
		Size: Discrete 640x480

./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: JPEG
    ... will try to handle this by checking against supported formats. 
 Init v4L2 failed !! exit fatal 
 i: init_VideoIn failed

jacksonliam, 2018-01-15:
Sorry you are all having this problem. I can't find much info on why this is suddenly happening, I think perhaps some of the drivers in the newer kernels have changed to report they only support JPEG instead of MJPG. I can't seem to find much info about what the difference between the two is either. 

I don't have a camera like this, can someone with the issue try something for me? 

In the file mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c change line 329.
`case V4L2_PIX_FMT_MJPEG:`
To
`case V4L2_PIX_FMT_JPEG:`

And rebuild (run make again)

I'd at least expect a different error message...

jacksonliam, 2018-01-15:
Note: other code is treating the two formats as identical e.g. https://webrtc-codereview.appspot.com/1120006/diff/1/webrtc/modules/video_capture/linux/video_capture_linux.cc#newcode160

mijofa, 2018-01-16:
Tried that, didn't work but did get a different error:
```
pi@octopi:~/mjpg-streamer-jacksonliam/mjpg-streamer-experimental $ sed -i '329s/_MJPEG/_JPEG/' plugins/input_uvc/v4l2uvc.c
pi@octopi:~/mjpg-streamer-jacksonliam/mjpg-streamer-experimental $ make
[...]
Scanning dependencies of target input_uvc
make[3]: Leaving directory '/home/pi/mjpg-streamer-jacksonliam/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer-jacksonliam/mjpg-streamer-experimental/_build'
[ 48%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/v4l2uvc.c.o
[ 52%] Linking C shared library input_uvc.so
make[3]: Leaving directory '/home/pi/mjpg-streamer-jacksonliam/mjpg-streamer-experimental/_build'
[ 64%] Built target input_uvc
[...]
pi@octopi:~/mjpg-streamer-jacksonliam/mjpg-streamer-experimental $ LD_LIBRARY_PATH=. ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -" -o "output_http.so -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: JPEG
    ... will try to handle this by checking against supported formats.
    ... Falling back to the faster MJPG mode (consider changing cmd line options).
 should never arrive exit fatal !!
 i: init_VideoIn failed
```

I figured I'd just try replacing all occurences of "MJPEG" with "JPEG" in that file, and that actually made it work. I don't expect this is actually suitable though as it probably makes MJPEG cameras work less efficiently, but figured it was worth testing in my case anyway:

```
pi@octopi:~/mjpg-streamer-jacksonliam/mjpg-streamer-experimental $ sed -i 's/_MJPEG/_JPEG/' plugins/input_uvc/v4l2uvc.c
pi@octopi:~/mjpg-streamer-jacksonliam/mjpg-streamer-experimental $ make
[...]
Scanning dependencies of target input_uvc
make[3]: Leaving directory '/home/pi/mjpg-streamer-jacksonliam/mjpg-streamer-experimental/_build'
make[3]: Entering directory '/home/pi/mjpg-streamer-jacksonliam/mjpg-streamer-experimental/_build'
[ 48%] Building C object plugins/input_uvc/CMakeFiles/input_uvc.dir/v4l2uvc.c.o
[ 52%] Linking C shared library input_uvc.so
make[3]: Leaving directory '/home/pi/mjpg-streamer-jacksonliam/mjpg-streamer-experimental/_build'
[ 64%] Built target input_uvc
[...]
pi@octopi:~/mjpg-streamer-jacksonliam/mjpg-streamer-experimental $ LD_LIBRARY_PATH=. ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -" -o "output_http.so -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: JPEG
    ... will try to handle this by checking against supported formats.
    ... Falling back to the faster MJPG mode (consider changing cmd line options).
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
### [ I manually tested the HTTP stream, it worked fine. ]
^Csetting signal to stop
 i: cleaning up resources allocated by input thread
force cancellation of threads and cleanup resources
 o: cleaning up resources allocated by server thread #00
done
```

jacksonliam, 2018-01-16:
Thanks, that helps actually! Will put together a fix, hopefully this weekend.

Ruthalas, 2018-01-21:
I have this issue as well! I am excited to see that a solution is on the horizon.

Jan2220, 2018-01-23:
Have the same issue but editing the v4l2uvc.c does not change anything for me... Still the same error
```
pi@octopi:~/mjpg-streamer/mjpg-streamer-experimental $ ./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: JPEG
    ... will try to handle this by checking against supported formats.
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed

```
Hope you can help me :/

jacksonliam, 2018-01-25:
I've pushed a fix for this JPEG problem. do a `git pull` then a `make` to build the latest code.

@Jan2220 Your LED problem is different, you could try merging in this commit https://github.com/jacksonliam/mjpg-streamer/pull/92 to see if it helps. But its not always possible to turn of the LED for USB webcams, so unless you've had success with it with other v4l apps I wouldn't hold your breath... (Small amount of black duct tape / electrical tape works for disabling LEDs!)

closed 2018-01-25 by jacksonliam

mijofa, 2018-01-26:
Thanks, much appreciated. I won't be able to test this until next week though. I'llo let you know how it goes.

nick5413, 2018-01-26:
I ran into this issue just now, using octopi with an older webcam.

I've pulled the latest version, built and installed it.  It works great!  Thank you for fixing this. :+1:

kaoss87, 2019-01-24:
> Sorry you are all having this problem. I can't find much info on why this is suddenly happening, I think perhaps some of the drivers in the newer kernels have changed to report they only support JPEG instead of MJPG. I can't seem to find much info about what the difference between the two is (in v4l2 terms) either.
> 
> I don't have a camera like this, can someone with the issue try something for me?
> 
> In the file mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c change line 329.
> `case V4L2_PIX_FMT_MJPEG:`
> To
> `case V4L2_PIX_FMT_JPEG:`
> 
> And rebuild (run make again)
> 
> I'd at least expect a different error message...

hi how to editing this file ? pls write the all code i am new the linux

jacksonliam, 2019-01-24:
> hi how to editing this file ? pls write the all code i am new the linux

You don't need to edit any files, this has been fixed.

kaoss87, 2019-01-24:
on case i use pc on rasbian not raspberry pi  pc spec vore 2 duo 1 gb ram etc

jacksonliam, 2019-01-24:
You have a different problem. Your camera does not support MJPG.

Try the -y option.

jacksonliam, 2019-01-24:
Please start a new issue for your camera and post the output of 
v4l2-ctl --list-formats-ext

md-rubel, 2021-01-29:
Having problem with H264 camera:

pi@raspberrypi:~/mjpg-streamer-1.0.0/mjpg-streamer-experimental $ lsusb
Bus 001 Device 003: ID 040d:300a VIA Technologies, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~/mjpg-streamer-1.0.0/mjpg-streamer-experimental $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'H264' (H.264, compressed)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 2048x1024
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
pi@raspberrypi:~/mjpg-streamer-1.0.0/mjpg-streamer-experimental $ ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -" -o "output_http.so -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: H264
    ... will try to handle this by checking against supported formats. 
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
pi@raspberrypi:~/mjpg-streamer-1.0.0/mjpg-streamer-experimental $

md-rubel, 2021-01-29:
pi@raspberrypi:~ /mjpg-streamer-1.0.0/mjpg-streamer-experimental $ lsusb
Bus 001 Device 003: ID 040d:300a VIA Technologies, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ /mjpg-streamer-1.0.0/mjpg-streamer-experimental $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'H264' (H.264, compressed)
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
		Size: Discrete 2048x1024
			Interval: Discrete 0.033s (30.000 fps)
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.100s (10.000 fps)
pi@raspberrypi:~ /mjpg-streamer-1.0.0/mjpg-streamer-experimental $ ./mjpg_streamer -i "input_uvc.so -d /dev/video0 -" -o "output_http.so -w ./www"
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: H264
    ... will try to handle this by checking against supported formats. 
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
pi@raspberrypi:~ /mjpg-streamer-1.0.0/mjpg-streamer-experimental $ 

[image]

commit 8cc9d22c (2018-01-25): Fix for webcams that return JPEG instead of MJPEG. Fixes #101