jacksonliam/mjpg-streamer #103: Taking over the orginal sourceforge project site

Torxgewinde, 2017-12-17:
Hello Jackson,
Some sites still point to the old sourceforge site. I havent updated or contributed for ages myself due to various private reasons. Would you be interested in syncing SF and this github or somehow make it point to this github site? I can offer to grant you admin rights to the project if you are interested. I can imagine that it would cause less confusion for users who simply try find a well maintained successor.

Kind regards,
Tom

jacksonliam, 2017-12-18:
Hey, thanks for getting in touch! 

I don't feel like syncing the source would be worth the effort but would be happy if you want to put a link here from the sourceforge site as a more recently maintained fork!

Cheers, 
Liam.

Torxgewinde, 2017-12-18:
Thanks Liam,
I will figure it out and make it point to github. Besides: Good job and thanks!

Best regards,
Tom

closed 2018-01-19 by jacksonliam

Torxgewinde, 2018-01-22:
Hello Liam,
I adjusted now the sourceforge site and it now points to your site. Thanks for all the effort you put into this and good luck!

Kind regards,
Tom

jacksonliam, 2018-01-22:
Great! Thanks!