jacksonliam/mjpg-streamer #105: configuring incomplete, errors occurred

AnduriI, 2018-01-15:
Hey there,

I tried to install mjpg-streamer to use it for my OctoPrint server. At the first try I had problems of cmake not finding OpenCV. I kind of resolved it by commenting out the input plugin line in CMakeLists.txt
After several tries removing and redownloading different forks of this project I'm out of ideas. I hope someone can help me find a solution. Here my output:
```
pi@OctoPi:~/mjpg-streamer/mjpg-streamer-experimental $ sudo make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
CMake Error: Error in cmake code at
/usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:198:
Parse error.  Expected a command name, got unquoted argument with text "0".
CMake Error at /usr/share/cmake-3.6/Modules/FindPkgConfig.cmake:48 (include):
  include could not find load file:

    /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake
Call Stack (most recent call first):
  cmake/FindGphoto2.cmake:21 (include)
  plugins/input_ptp2/CMakeLists.txt:2 (find_package)

CMake Error at /usr/share/cmake-3.6/Modules/FindPkgConfig.cmake:49 (find_package_handle_standard_args):
  Unknown CMake command "find_package_handle_standard_args".
Call Stack (most recent call first):
  cmake/FindGphoto2.cmake:21 (include)
  plugins/input_ptp2/CMakeLists.txt:2 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build/CMakeFiles/CMakeOutput.log".
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 1
```
I also had a look at the CMakeOutput.log, but didn't found any suspicious line. In case it will offer information to you I can upload it as well.

jacksonliam, 2018-01-15:
It should work without opencv, cmake should detect it is not installed, then continue without.

Can you show me the error with opencv? 

What OS is this? E.g. `uname -a`.

AnduriI, 2018-01-16:
I did retry, here is the output, hope it helps:
```
pi@OctoPi:~ $ sudo rm -r mjpg-streamer/
pi@OctoPi:~ $ git clone https://github.com/jacksonliam/mjpg-streamer.git
Klone nach 'mjpg-streamer'...
remote: Counting objects: 2655, done.
remote: Total 2655 (delta 0), reused 0 (delta 0), pack-reused 2655
Empfange Objekte: 100% (2655/2655), 3.43 MiB | 2.78 MiB/s, Fertig.
Löse Unterschiede auf: 100% (1664/1664), Fertig.
Prüfe Konnektivität... Fertig.
pi@OctoPi:~ $ cd mjpg-streamer/mjpg-streamer-experimental
pi@OctoPi:~/mjpg-streamer/mjpg-streamer-experimental $ export LD_LIBRARY_PATH=.
pi@OctoPi:~/mjpg-streamer/mjpg-streamer-experimental $ make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMAKE_BUILD_TYPE = Release
-- Looking for include file sys/inotify.h
-- Looking for include file sys/inotify.h - found
CMake Warning at plugins/input_opencv/CMakeLists.txt:4 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Error: Error in cmake code at
/usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:198:
Parse error.  Expected a command name, got unquoted argument with text "0".
CMake Error at /usr/share/cmake-3.6/Modules/FindPkgConfig.cmake:48 (include):
  include could not find load file:

    /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake
Call Stack (most recent call first):
  cmake/FindGphoto2.cmake:21 (include)
  plugins/input_ptp2/CMakeLists.txt:2 (find_package)

CMake Error at /usr/share/cmake-3.6/Modules/FindPkgConfig.cmake:49 (find_package_handle_standard_args):
  Unknown CMake command "find_package_handle_standard_args".
Call Stack (most recent call first):
  cmake/FindGphoto2.cmake:21 (include)
  plugins/input_ptp2/CMakeLists.txt:2 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build/CMakeFiles/CMakeOutput.log".
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 1
```

jacksonliam, 2018-01-18:
Does removing the line `add_subdirectory(plugins/input_ptp2)` from the file CMakeLists.txt help?

AnduriI, 2018-01-18:
nope, removing ptp2 brings this output:
```
pi@OctoPi:~/mjpg-streamer/mjpg-streamer-experimental $ make
[ -d _build ] || mkdir _build
[ -f _build/Makefile ] || (cd _build && cmake  -DCMAKE_BUILD_TYPE=Release ..)
CMAKE_BUILD_TYPE = Release
CMake Warning at plugins/input_opencv/CMakeLists.txt:4 (find_package):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

CMake Error: Error in cmake code at
/usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake:198:
Parse error.  Expected a command name, got unquoted argument with text "0".
CMake Error at /usr/share/cmake-3.6/Modules/FindThreads.cmake:222 (include):
  include could not find load file:

    /usr/share/cmake-3.6/Modules/FindPackageHandleStandardArgs.cmake
Call Stack (most recent call first):
  /usr/share/cmake-3.6/Modules/FindSDL.cmake:133 (find_package)
  plugins/output_viewer/CMakeLists.txt:2 (find_package)

CMake Error at /usr/share/cmake-3.6/Modules/FindThreads.cmake:223 (FIND_PACKAGE_HANDLE_STANDARD_ARGS):
  Unknown CMake command "FIND_PACKAGE_HANDLE_STANDARD_ARGS".
Call Stack (most recent call first):
  /usr/share/cmake-3.6/Modules/FindSDL.cmake:133 (find_package)
  plugins/output_viewer/CMakeLists.txt:2 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build/CMakeFiles/CMakeError.log".
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 1
```
Another strange thing: it says FindPackageHandleStandardArgs.cmake has a problem at line 198. I just had a brief look to this file and it has a lonely `0` in this line... don't think that's correct...

```
function(FIND_PACKAGE_HANDLE_STANDARD_ARGS _NAME _FIRST_ARG)

# set up the arguments for CMAKE_PARSE_ARGUMENTS and check whether we are in
# new extended or in the "old" mode:
  set(options  CONFIG_MODE  HANDLE_COMPONENTS)
  set(oneValueArgs  FAIL_MESSAGE  VERSION_VAR  FOUND_VAR)
  set(multiValueArgs REQUIRED_VARS)
  set(_KEYWORDS_FOR_EXTENDED_MODE  ${options} ${oneValueArgs} ${multiValueArgs} )
  list(FIND _KEYWORDS_FOR_EXTENDED_MODE "${_FIRST_ARG}" INDEX)

  if(${INDEX} EQUAL -1)
0
^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@$
.endif

.ifdef MAIN_TLS_ENABLE
.ifndef MAIN_TLS_ADVERTISE_HOSTS
MAIN_TLS_ADVERTISE_HOSTS = *
.endif
tls_advertise_hosts = MAIN_TLS_ADVERTISE_HOSTS
```

jacksonliam, 2018-01-18:
Yeah I think your cmake installation is broken!

AnduriI, 2018-01-19:
I got it... it was not only cmake having problems. Just for completeness, I will post the solution here (in case someone has a similiar problem in future):
I tried to uninstall cmake with `sudo apt-get purge cmake` leading to an error saying `libtiff5-dev` has empty file lists (is that the correct englisch translation? In german it's "Dateilisten-Datei des Paketes »libtiff5-dev« enthält leeren Dateinamen")
I had a look at `/var/lib/dpkg/info/libtiff5-dev\:armhf.list and it contained not only folder and file names, but had also info about installed programm:
```
[...]
/usr/share/man/man3/TIFFbuffer.3tiff.gz
/usr/share/man/man3/TIFFClose.3tiff.gz
/usr/share/manng Python programs to easily perform a
 variety of functions.
 .
 This package contains locales.

Package: python-dnspython
Status: install ok installed
Priority: optional
Section: python
[...]
```
I did a backup and deleted those lines. After that I could use `sudo dpkg --configure -a` and go on as intended: purge cmake, autoremove all dependencies and reinstall. Now I can use `make` and the stream works well.

Thanks a lot @jacksonliam for your help.

closed 2018-01-19 by AnduriI