jacksonliam/mjpg-streamer #114: Could not obtain the requested pixelformat (RGB3)

Skystriker, 2018-02-24:
Hi, I got an issue with my webcam connecting to OctoPrint

The /var/log/webcamd.log displayes this:

logger: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i input_uvc.so -r$
MJPG Streamer Version.: 2.0
i: Using V4L2 device.: /dev/video0
i: Desired Resolution: 640 x 480
i: Frames Per Second.: 10
i: Format............: JPEG
i: TV-Norm...........: DEFAULT
i: The specified resolution is unavailable, using: width 320 height 240 instea$
i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3

Seems like it is seeing the webcam. But it is not outputting the correct filetype. Is there some way around this?

Cheers
Sky

jacksonliam, 2018-02-25:
Looks like your webcam doesn't support jpeg, try adding the -y flag to pull frames in yuv mode.

(note, this uses more cpu)

closed 2018-02-25 by jacksonliam

Skystriker, 2018-02-26:
Hi, sadly that doesnt help.

jacksonliam, 2018-02-26:
Can you post the output of running
`v4l2-ctl --list-formats-ext`?

jacob-swanson, 2018-04-15:
I'm having this same problem with a fairly old webcam. It works fine with Cheese.

```
$ ./mjpg_streamer -i "./input_uvc.so -y" -o "./output_http.so"
MJPG Streamer Version: git rev: 821c330ea6bbb5fbed98d48e00aac156e923161b
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: YUYV , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
 Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
```

```
$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 176x144
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 160x120
                        Interval: Discrete 0.033s (30.000 fps)
                        Interval: Discrete 0.040s (25.000 fps)
                        Interval: Discrete 0.050s (20.000 fps)
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.100s (10.000 fps)
                        Interval: Discrete 0.200s (5.000 fps)
                Size: Discrete 1280x1024
                        Interval: Discrete 0.111s (9.000 fps)
```

```
$ v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
        Index       : 0
        Type        : Video Capture
        Pixel Format: 'YUYV'
        Name        : YUYV 4:2:2
```

```
$ lsusb
Bus 001 Device 003: ID 064e:a116 Suyin Corp. UVC 1.3MPixel WebCam
```

jacksonliam, 2018-04-16:
Shot in the dark, but try providing one of the supported fps as an argument. 

It's weird that v4l2-ctl isn't even listing RGB3, yet it looks like that's the format the driver hands us :-/

What hardware / OS are you using? Did you build mjpg-streamer yourself?

jacob-swanson, 2018-04-16:
I tried providing the fps and resolution, but got the same result.

It's one of these https://www.newegg.com/Product/Product.aspx?Item=N82E16834220385. It's about 9 years old at this point. I'm trying to setup OctoPrint. It's not a big deal if I can't get it to work.

I pulled and built it yesterday,, and I'm using Xubuntu 16.04 LTS.

jacksonliam, 2018-04-16:
That should be OK then.

You could try the `-u` flag instead of the `-y` one. Also could try `-fourcc RGBP` instead of `-y`. 

It may also be OS, could try something more recent?

After that I'm not sure I have any more suggestions, would need hardware to be able to reproduce the issue.

jacob-swanson, 2018-04-17:
No luck. I tried with Xubuntu 17.10 too. Seems like it has some sort of fallback to RGB3 for the pixel format no mater what you try to specify.

jacksonliam, 2018-04-17:
Sorry about that, thanks for trying. Unfortunately I don't think I can help unless I can get my hands on a camera that has these symptoms!

jacob-swanson, 2018-04-18:
I don't think there's a problem with this project. I tried using VLC to stream the webcam and it had a similar error. The webcam is listed on the UVC website. 🤷‍♂️