jacksonliam/mjpg-streamer #118: Logging access / viewers

djkraven, 2018-03-10:
Is there an option or way to log the viewers of the stream? I have this running on RaspberryPi and LinuxMint and it would good to be able to display current viewers and track the IP's of the viewers. This would be helpful from a usage interest but also security to know if the stream is being abused.

jacksonliam, 2018-03-10:
It looks like if you build with http management enabled, you can do a GET /clients.json but I can't say I've used it. 

I _think_ to build with management enabled, you should be able to do an advanced build (see README.md) and then when you run `make`, instead run `make CFLAGS=-DMANAGEMENT`

But I can't say I've used it in a while, also management might enable some other stuff, but unfortunately there's no documentation of what that might be so you'd need to check the code!

djkraven, 2018-03-13:
I tried the make CFLAGS=-DMANAGEMENT but the /client.json returns a 404 error though the /program.json returns information. Trying check I didn't have a typo I looked at the Makefile and searched other files and didn't find a reference to the -DMANAGEMENT option.  Any suggestions?

jacksonliam, 2018-03-13:
Hmmm yes it doesn't seem to work even with management enabled. I'll rename this issue to cover that problem.

jacksonliam, 2018-04-04:
OK I've figured out how to build MANAGEMENT so that clients.json is enabled.

```
    cd mjpg-streamer-experimental
    mkdir _build
    cd _build
    cmake -DENABLE_HTTP_MANAGEMENT=ON ..
    cd ..
    make
```

clients.json looks like this:
```
{
"clients": [
{
"address": "192.168.1.55",
"timestamp": 1522866831
}
]
}
```

closed 2018-04-04 by jacksonliam