jacksonliam/mjpg-streamer #121: Please Add A Config File

vicimikec, 2018-03-20:
It would make running mjpg-streamer as a service considerably easier if it had a config file. Currently to change an option I have to edit the systemd unit file, reload the systemd unit files, then restart the service. This gets cumbersome when trying to find the right options for a specific camera. With a config file it would be just a matter of changing the config and restarting the service. Something I think far more people are comfortable doing.

vicimikec, 2018-03-21:
I figured out a way in systemd to do this. Attached are the /etc/default/mjpg_streamer and /lib/systemd/system/mjpg_streamer.service files I am using. GitHub would not let me upload them without .txt on the end so you will need to remove those. Also, for some reason my system installed them the exec in /usr/local/bin which seems odd. Might need to adjust this.

[mjpg_streamer.txt](https://github.com/jacksonliam/mjpg-streamer/files/1831984/mjpg_streamer.txt)
[mjpg_streamer.service.txt](https://github.com/jacksonliam/mjpg-streamer/files/1831985/mjpg_streamer.service.txt)

jacksonliam, 2018-03-21:
Thanks, that's a good solution, I'll look to pull these in