jacksonliam/mjpg-streamer #122: Compilation on OSX?

jeffh0821, 2018-03-22:
Any insight, successes, or outright ridicule for trying to compile this for OSX? The AVFoundation library for OSX doesn't allow much in the way of accessing the MJPEG formats within UVC cameras and leveraging mjpg-streamer would be a handy workaround. I gave it a whirl with cmake and libjpeg available from homebrew but it bombed out when attempting to compile - looking for the videodev2 headers which aren't there.

I totally accept that it probably isn't possible - but thought I would ask anyway....

jacksonliam, 2018-03-22:
Does osx have v4l2 support? If not you'd probably need an osx specific input plugin, rather than trying to use the uvc one. 

cmake should figure out v4l2 is not available and skip building the uvc plugin, I may need to look into why that doesn't seem to be happening here.

jeffh0821, 2018-03-22:
Unfortunately, OSX does not leverage v4l2. It has a generic UVC driver and, typically, access to webcams is done through their own AVFoundation library. The problem is the AVFoundation library gives access to almost no controls in the UVC plugin and you can only access the raw video (YUYV). Cannot pull the MJPEG video with the higher framerates.

I may have to stop being so lazy and just hang the UVC camera (Logitech QuickCam Pro 9000) off a Pi and use ffmpeg or mjpg-streamer there. But I'm a sucker for a challenge...

jacksonliam, 2018-10-21:
I don't think either of us are going to make any progress on this 😄

closed 2018-10-21 by jacksonliam