jacksonliam/mjpg-streamer #123: Could not launch another client thread

sluebbert, 2018-03-27:
Mjpg-streamer seems to run out of resources or threads eventually. I have it running on arch for raspberry pi.  Motion is installed and running on an in-home server and constantly streaming from the raspberry pi feed. No other devices connect to the pi's video stream.

The following command is setup as a systemd service:
mjpg_streamer -o output_http.so -w ./www -i input_raspicam.so -x 1920 -y 1080 -fps 2 -quality 98 -usestills

After several hours, mjpg-streamer will stop serving clients. I can't get it to respond on its port. It will fall to the bottom of the stack in resource usage where it typically starts out at the top. For example, it is currently running with 0% CPU and 1.8% memory.  It also typically starts out with using 11 threads and ends with around 251 even though the limit for the process is about 843. I have no idea when the thread count starts climbing.

After kicking the thing into debug mode, eventually it complains over and over about "could not launch another client thread".  I tweaked it to log the result code too which happens to be 11 or EAGAIN.

Around the beginning (no issue) (constantly happening every second except for the 100 frames line):
`
.../output_http/httpd.c, send_stream(), 508): got frame (size: 272 kB)  
.../output_http/httpd.c, send_stream(), 525): sending intemdiate header  
.../output_http/httpd.c, send_stream(), 528): sending frame  
.../output_http/httpd.c, send_stream(), 531): sending boundary  
.../output_http/httpd.c, send_stream(), 508): got frame (size: 405 kB)  
.../output_http/httpd.c, send_stream(), 525): sending intemdiate header  
.../output_http/httpd.c, send_stream(), 528): sending frame  
.../output_http/httpd.c, send_stream(), 531): sending boundary  
.../output_http/httpd.c, send_stream(), 508): 100 frames captured in 32.846868 seconds (3.044430 fps)  

`

Around the issue (repeats happen exactly every 25 seconds):
`
.../output_http/httpd.c, server_thread(), 1574): create thread to handle client that just established a connection  
.../output_http/httpd.c, server_thread(), 1577): serving client: 192.168.1.103  
.../output_http/httpd.c, server_thread(), 1546): waiting for clients to connect  
.../output_http/httpd.c, client_thread(), 1255): plugin_no: 0  
.../output_http/httpd.c, client_thread(), 1317): Request for stream from input: 0  
.../output_http/httpd.c, send_stream(), 465): preparing header  
.../output_http/httpd.c, send_stream(), 478): Headers send, sending stream now  
.../output_http/httpd.c, server_thread(), 1574): create thread to handle client that just established a connection  
.../output_http/httpd.c, server_thread(), 1577): serving client: 192.168.1.103  
.../output_http/httpd.c, server_thread(), 1546): waiting for clients to connect  
.../output_http/httpd.c, client_thread(), 1255): plugin_no: 0  
.../output_http/httpd.c, client_thread(), 1317): Request for stream from input: 0  
.../output_http/httpd.c, send_stream(), 465): preparing header  
.../output_http/httpd.c, send_stream(), 478): Headers send, sending stream now  

`

Well after the issue (repeats happen exactly every 5 seconds):
`
.../output_http/httpd.c, server_thread(), 1546): waiting for clients to connect  
.../output_http/httpd.c, server_thread(), 1574): create thread to handle client that just established a connection  
.../output_http/httpd.c, server_thread(), 1577): serving client: 192.168.1.103  
.../output_http/httpd.c, server_thread(), 1586): could not launch another client thread: 11  
.../output_http/httpd.c, server_thread(), 1546): waiting for clients to connect  
.../output_http/httpd.c, server_thread(), 1574): create thread to handle client that just established a connection  
.../output_http/httpd.c, server_thread(), 1577): serving client: 192.168.1.103  
.../output_http/httpd.c, server_thread(), 1586): could not launch another client thread: 11  
.../output_http/httpd.c, server_thread(), 1546): waiting for clients to connect  

`
These log dumps represent changes. - As in the first dump happens over and over for a long time, the second dump happens over and over for a short time, and the final dump happens over and over until reset.  I didn't see anything else around their boundaries.

I'm guessing that 25 second timeouts are happening and causing issues. I don't know why the raspberry pi and in-home server would be getting timeouts though. They are connected through a router by ethernet. I haven't looked into that though.

Any thoughts?

Thanks.

jacksonliam, 2018-03-27:
Could the client be keeping connections open but for some reason not reusing them,  instead starting new ones, essentially DOS-ing mjpg-streamer?

As a workaround can you use the snapshot rather than stream URI on the client?

jacksonliam, 2018-03-28:
Agreed, unless motion is keeping the connections open, the threads should terminate. 

Which is what is making me think motion is not closing the connection or just taking a long time to do so (long timeout).

sluebbert, 2018-03-29:
I haven't had any issues with threads with the static image setup; at least not yet. Mjpg-streamer is still holding steady now with 11 threads even 24 hours later.  The steaming setup would have runaway threads after just a few hours.

Killing the client did not impact or fix mjpg-streamer when I was troubleshooting earlier.  It seems to stay in a faulted state until reset.

sluebbert, 2018-04-25:
I haven't been able to focus on this in a while, but it looks like I am getting the same issues with using action=snapshot. At some point the thread count shoots up to 251 and no new connections can be stabilized. Mjpg-streamer must be restarted before normal activity can continue. Turning off motion after the issue does not help.

I'll try setting motion to close the socket after each http request to see if that helps. I have been using motion with the keepalive header enabled.