jacksonliam/mjpg-streamer #124: input_uvc.so

prahjister, 2018-03-30:
Is it possible to add a timestamp to this....looks like it is only implemented in raspicam

jacksonliam, 2018-03-30:
Exif or http header?

prahjister, 2018-03-30:
Can you do it exactly like raspicam. I asked mr dewitt shown above.

jhdewitt, 2018-04-04:
Referring to: https://github.com/jacksonliam/mjpg-streamer/commit/1b22b2e2e4194d3145628e07659b7c848cd8ae1d

`pglobal->in[plugin_number].timestamp = timestamp;`

This is how I extract the timestamp from byte stream (python):
```
    t = bytes.find('Timestamp')
    t2 = t+bytes[t:t+40].find(".")
    a = bytes.find('\xff\xd8')
    b = bytes.find('\xff\xd9')
    if a!=-1 and b!=-1:
       tstamp_s = bytes[t+11:t2]
       tstamp_ss = bytes[t2+1:t2+7]
       frame_time = float(tstamp_s) + float('0.'+tstamp_ss)
```

Is this http header then?

jhdewitt, 2018-04-04:
Would this be the appropriate place to sample time?

https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c#L632

```
/* grab a frame */
        if(uvcGrab(pcontext->videoIn) < 0) {
            IPRINT("Error grabbing frames\n");
            exit(EXIT_FAILURE);
}
```

jacksonliam, 2018-04-04:
Yes that's the http header.

Here would be a better place to do the sampling 
https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c#L710

prahjister, 2018-04-07:
I tested the proped changes and it didn't compile

jacksonliam, 2018-04-14:
The changes are merged, thanks to @jhdewitt !

closed 2018-04-14 by jacksonliam