jacksonliam/mjpg-streamer #125: How can I use rstp

ighack, 2018-03-31:
mjpg_streamer -i "input_uvc.so -y" -o "output_rtsp.so -p 8081"
and I via VLC rtsp://pi_of_ip:8081.
I can't see anything
what's solution

PS:
I can see "mjpg_streamer -i "input_uvc.so -y"  -o "output_http.so -w www" " with rtsp://pi_of_ip:8080

rhobincu, 2018-06-08:
The RTSP plugin is not functional. All it does is expect a filename over UDP and saves a frame from the source as that name. It does not implement the RSTP protocol at all. 

Also, technically RTSP is a control protocol, not a transport one, that means that it is used only to control streaming (play, pause, rewind), not actually transfer video data to the client. That is done using the RTP protocol that is not implemented either in mjpg_streamer.

cms-, 2018-06-08:
The UDP plugin is nearly identical.

jacksonliam, 2018-10-21:
Documentation has been update to note functionality.

closed 2018-10-21 by jacksonliam