jacksonliam/mjpg-streamer #126: Stopped working

qpens, 2018-04-03:
Hi,
I'm running Ubuntu 16.04.4 LTS (Xenial Xerus) on Odroid C2 after reinstalling OS I'm getting an error.

[image]
 
My installation steps:
sudo apt-get install cmake subversion libjpeg62-dev imagemagick libav-tools libv4l-dev 
git clone https://github.com/jacksonliam/mjpg-streamer.git
cd mjpg-streamer/mjpg-streamer-experimental
export LD_LIBRARY_PATH=.
make

./mjpg_streamer -i "./input_uvc.so -r HD -f 10" -o "./output_http.so"

jacksonliam, 2018-04-04:
Those ioctl errors should be harmless. 

I assume you cannot see the stream?

Can you try with 30fps? Your camera doesn't actually support a 10fps mode.

jacksonliam, 2018-04-04:
Were you running the same OS version before the reinstall or have you upgraded?

I think you are seeing this issue: https://github.com/jacksonliam/mjpg-streamer/issues/53

qpens, 2018-04-04:
I can't compile his older version. Is there any other source to get a working version?

jacksonliam, 2018-04-04:
Yeah old versions of mjpg-streamer dont work with newer kernels. 

Are you sure it was this fork of mjpg-streamer you were using before the reinstall?

Were you running the same OS version before the reinstall or have you upgraded?

qpens, 2018-04-04:
yes

qpens, 2018-04-04:
If that helps. 
https://wiki.odroid.com/odroid-c2/os_images/ubuntu/v2.4

mad-ady, 2018-04-05:
I think the main difference is that Odroid C2 will move to kernel 3.16 (instead of 3.14) and on 3.16 mjpg-streamer stops working. On 3.14 all is well.
Here is a discussion on the odroid forum about this issue: https://forum.odroid.com/viewtopic.php?f=136&t=30658

@jacksonliam: Is there any debug we can enable (apart from strace) that could tell us what kernel features it tries to use? There may be some differences in Kernel 3.16 compared to 3.14.

mad-ady, 2018-04-05:
Update: Once I start mjpg_streamer with strace I get loads of these calls:
[pid 30955] ioctl(4, VIDIOC_DQBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=0, memory=V4L2_MEMORY_MMAP, m.offset=0, length=1843789, bytesused=73288, flags=V4L2_BUF_FLAG_MAPPED|0x12000, timestamp = {1581.303627}, ...}) = 0
[pid 30955] ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=0, memory=V4L2_MEMORY_MMAP, m.offset=0, length=1843789, bytesused=0, flags=V4L2_BUF_FLAG_MAPPED|V4L2_BUF_FLAG_QUEUED|0x12000, ...}) = 0
[pid 30955] ioctl(4, VIDIOC_DQBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=1, memory=V4L2_MEMORY_MMAP, m.offset=0x1c3000, length=1843789, bytesused=72808, flags=V4L2_BUF_FLAG_MAPPED|0x12000, timestamp = {1581.335782}, ...}) = 0
[pid 30955] ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=1, memory=V4L2_MEMORY_MMAP, m.offset=0x1c3000, length=1843789, bytesused=0, flags=V4L2_BUF_FLAG_MAPPED|V4L2_BUF_FLAG_QUEUED|0x12000, ...}) = 0
[pid 30955] ioctl(4, VIDIOC_DQBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=2, memory=V4L2_MEMORY_MMAP, m.offset=0x386000, length=1843789, bytesused=72904, flags=V4L2_BUF_FLAG_MAPPED|0x12000, timestamp = {1581.371691}, ...}) = 0
[pid 30955] ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=2, memory=V4L2_MEMORY_MMAP, m.offset=0x386000, length=1843789, bytesused=0, flags=V4L2_BUF_FLAG_MAPPED|V4L2_BUF_FLAG_QUEUED|0x12000, ...}) = 0
[pid 30955] ioctl(4, VIDIOC_DQBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=3, memory=V4L2_MEMORY_MMAP, m.offset=0x549000, length=1843789, bytesused=73072, flags=V4L2_BUF_FLAG_MAPPED|0x12000, timestamp = {1581.403882}, ...}) = 0
[pid 30955] ioctl(4, VIDIOC_QBUF, {type=V4L2_BUF_TYPE_VIDEO_CAPTURE, index=3, memory=V4L2_MEMORY_MMAP, m.offset=0x549000, length=1843789, bytesused=0, flags=V4L2_BUF_FLAG_MAPPED|V4L2_BUF_FLAG_QUEUED|0x12000, ...}) = 0

qpens, 2018-04-05:
I made some screens with debugging.
[image]
[image]

mad-ady, 2018-04-11:
I have found a partial fix, or at least the cause of the problem.
It seems that mjpg_streamer works on kernel 3.16 **only if you match the requested framerate (-f) with the supported framerate by the camera (v4l2-ctl --list-formats-ext).**
In my case, for 720p the camera supports only:

>                     Size: Discrete 1280x720                                         
>                             Interval: Discrete 0.033s (30.000 fps)                 
>                             Interval: Discrete 0.067s (15.000 fps)

and I was asking for 25 fps, which caused it to stall on this kernel. If I remove -f or if I ask for -f 15 or -f 30, then the program works as expected.

I can see that the program outputs  _i: FPS coerced ......: from 25 to 30_ when there's a mismatch, but in this case the image stream doesn't work. Any ideas if there needs to be something enabled in the kernel, or if it's a bug somewhere?

jacksonliam, 2018-04-11:
This was my thinking earlier, but @qpens said that using 30 fps made no difference.

The software frame dropping/coercive  code is a bit messy (I didn't write it!) so it could well be related. I'm tempted to remove it.

jacksonliam, 2018-04-11:
Basically if the camera gives us a different fps to the one we asked for, we set it again to 255 to get the camera to run as high an fps as possible, so when we drop frames (to get down to the requested framerate) the stream won't be so juddery.

However, I recon some cameras/drivers (possibly a driver update in a newer kernel) have a problem either with either the FPS being set twice, or being set to 255.

But I think the whole notion of automatic frame dropping is silly, and people should just get what fps their camera gives them (with frame dropping being a separate command line option).

But if people are trying the correct framerate (that their camera supports) like I ask them, and report that they are still seeing no stream, then there is something else going on.

qpens, 2018-04-11:
I made some tests and when trying resolution 1920x1080 fps 30 my camera start working when dropping to 25fps it crushes after that I change resolution to 1280x720 fps 30 and magically it worked dropped to 25 crash again. Other thing is that even if fps is 30 I'm getting only 5fps

mad-ady, 2018-04-12:
@qpens: the low framerate you're seeing might be caused by problems in the
USB bus - I sometimes get corrupted frames from the camera on the C2, but
not on other systems. It's a known limitation of the C2's  USB
implementation.

jacksonliam, 2018-04-14:
I have pushed changes I think fixed this (821c330ea6bbb5fbed98d48e00aac156e923161b) can you please retry?
e.g.
```git pull
make distclean
make```
(and if you ran it before)
```make install```

Cheers.

The problem was related to the software framedropping, but it was because the framedropping code was not using the timestamp we copy into userspace, instead using one which is reclaimed/cleared by the driver.

qpens, 2018-04-14:
Works! Thanks

closed 2018-04-14 by jacksonliam