jacksonliam/mjpg-streamer #128: cvfilter_cpp.so doesn't work

a-loner, 2018-04-07:
The default **cvfilter_cpp.so** cannot do anything except copying original image (i.e. dst=src). Specifically, even just adding a simple drawing function would result in error. 

# Problem

**file:**  /home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/filters/cvfilter_cpp/filter_cpp.cpp

**Original**
```cpp
void filter_process(void* filter_ctx, Mat &src, Mat &dst) {
    // TODO insert your filter code here
    dst = src;
}
```  

**Add drawing function**
```cpp
void filter_process(void* filter_ctx, Mat &src, Mat &dst) {
    // TODO insert your filter code here
    circle( src, Point( 200, 200 ), 32.0, Scalar( 0, 0, 255 ), 1, 8 );   // add this line
    dst = src;
}
```
**Error**
```shell
pi@raspberrypi:~/mjpg-streamer/mjpg-streamer-experimental $ ./build/mjpg_streamer -o "output_http.so -w ./www" -i "input_opencv.so -r 1280x480 --filter cvfilter_cpp.so"
MJPG Streamer Version: git rev: 8cc9d22c1e79905d529a248ccf05bbf0625e0bf3
 i: device........... : default
 i: Desired Resolution: 1280 x 480
 i: filter........... : cvfilter_cpp.so
 i: filter args ..... : 
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
select timeout
./build/mjpg_streamer: symbol lookup error: /usr/local/lib/mjpg-streamer/cvfilter_cpp.so: undefined symbol: _ZN2cv6circleERKNS_17_InputOutputArrayENS_6Point_IiEEiRKNS_7Scalar_IdEEiii
```
# Solution
add **link_libraries(${OpenCV_LIBS})** in file **/home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/CMakeLists.txt** as below
```cmake
# TODO: which components do I need?
# To fix the error: "undefined symbol: _ZN2cv12VideoCaptureC1Ev"
find_package(OpenCV COMPONENTS core imgproc highgui videoio)

MJPG_STREAMER_PLUGIN_OPTION(input_opencv "OpenCV input plugin"
                            ONLYIF OpenCV_FOUND)

if (PLUGIN_INPUT_OPENCV)
    include_directories(${OpenCV_INCLUDE_DIRS})

    MJPG_STREAMER_PLUGIN_COMPILE(input_opencv input_opencv.cpp)
    
    link_libraries(${OpenCV_LIBS})

    target_link_libraries(input_opencv ${OpenCV_LIBS})
    
    add_subdirectory(filters/cvfilter_cpp)
    add_subdirectory(filters/cvfilter_py)
    
endif()
```

jacksonliam, 2018-04-07:
Thanks for the detailed report, I'll look into getting this merged

NozomiWyane, 2018-07-02:
How to Link my own library and include?