jacksonliam/mjpg-streamer #130: mjpg-streamer crash on OctoPi running on a Pi 3 B+

cassianoleal, 2018-04-15:
I'm running Octoprint on a Pi 3 B+, everything up-to-date.

mjpg-streamer seems to cause a Kernel crash. Everything seems to be working fine but this happens every so often and I get a bunch of rubbish on my SSH screen.

The webcam being streamed is a [GUCEE HD92 Web Camera](https://www.amazon.co.uk/gp/product/B01NAKVPDJ/), which uses the V4L/V4L2 drivers.

```Apr 15 17:34:04 octopi kernel: [ 1436.638218] Unable to handle kernel NULL pointer dereference at virtual address 00000028
Apr 15 17:34:04 octopi kernel: [ 1436.642354] pgd = b313c000
Apr 15 17:34:04 octopi kernel: [ 1436.644402] [00000028] *pgd=34c49835, *pte=00000000, *ppte=00000000
Apr 15 17:34:04 octopi kernel: [ 1436.646517] Internal error: Oops: 17 [#5] SMP ARM
Apr 15 17:34:04 octopi kernel: [ 1436.648561] Modules linked in: ftdi_sio usbserial ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink iptable_nat nf_conntrack_
ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter xt_conntrack nf_nat nf_conntrack br_netfilter bridge stp llc overlay xt_tcpudp tun iptable_mangle xt_DSCP cmac bnep hci_uart bluetooth ecdh_generic evdev brcmfmac fb_hx8357d(C) fbtft(C) brcmutil syscopyarea sysfillrect sysimgblt fb_sys_fops cfg80211 rfkill snd_bcm2835(C) gpio_backlight i2c_bcm2835 spi_bcm2835 uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_usb_audio snd_hwdep snd_usbmidi_lib snd_rawmidi snd_seq_device videobuf2_core snd_pcm snd_timer videodev snd media uio_pdrv_genirq fixed uio ip_tables x_tables ipv6
Apr 15 17:34:04 octopi kernel: [ 1436.670537] CPU: 2 PID: 1422 Comm: mjpg_streamer Tainted: G      D  C      4.14.30-v7+ #1102
Apr 15 17:34:04 octopi kernel: [ 1436.675093] Hardware name: BCM2835
Apr 15 17:34:04 octopi kernel: [ 1436.677314] task: b3129e00 task.stack: adb2e000
Apr 15 17:34:04 octopi kernel: [ 1436.679580] PC is at tcp_push+0x44/0x148
Apr 15 17:34:04 octopi kernel: [ 1436.681816] LR is at tcp_sendmsg_locked+0x308/0xdfc
Apr 15 17:34:04 octopi kernel: [ 1436.684007] pc : [<806ce384>]    lr : [<806d1950>]    psr: 40000013
Apr 15 17:34:04 octopi kernel: [ 1436.686191] sp : adb2fd78  ip : b326b2fc  fp : adb2fd9c
Apr 15 17:34:04 octopi kernel: [ 1436.688405] r10: adb03080  r9 : ffffffe0  r8 : b326b2fc
Apr 15 17:34:04 octopi kernel: [ 1436.690603] r7 : 00000000  r6 : 00000abc  r5 : 00000000  r4 : b326b200
Apr 15 17:34:04 octopi kernel: [ 1436.692798] r3 : 00000000  r2 : 00000abc  r1 : 00000000  r0 : a3966480
Apr 15 17:34:04 octopi kernel: [ 1436.694927] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Apr 15 17:34:04 octopi kernel: [ 1436.697052] Control: 10c5383d  Table: 3313c06a  DAC: 00000055
Apr 15 17:34:04 octopi kernel: [ 1436.699204] Process mjpg_streamer (pid: 1422, stack limit = 0xadb2e210)
Apr 15 17:34:04 octopi kernel: [ 1436.701374] Stack: (0xadb2fd78 to 0xadb30000)
Apr 15 17:34:04 octopi kernel: [ 1436.703519] fd60:                                                       0000f6e4 806d1930
Apr 15 17:34:04 octopi kernel: [ 1436.707669] fd80: 0001a934 adb2fe88 b326b200 0001a934 adb2fe34 adb2fda0 806d1950 806ce34c
Apr 15 17:34:04 octopi kernel: [ 1436.711761] fda0: 0000f6e4 adb2fe6c adb2fdcc adb2fdb8 80c069c4 80d0a08c 80c049c0 adb2e020
Apr 15 17:34:04 octopi kernel: [ 1436.715866] fdc0: 00000000 b312a45c 00000001 00000000 00000001 00000000 00010000 00000000
Apr 15 17:34:04 octopi kernel: [ 1436.719984] fde0: 00000002 b326b350 00000abc 00000000 00000000 0000f6e4 7fffffff adb2fe08
Apr 15 17:34:04 octopi kernel: [ 1436.724112] fe00: 801235c4 801e0000 b326b200 b326b200 00024f58 adb2fe88 00000000 adb2ff08
Apr 15 17:34:04 octopi kernel: [ 1436.728274] fe20: adafc9c0 00000000 adb2fe54 adb2fe38 806d247c 806d1654 806d2444 b326b200
Apr 15 17:34:04 octopi kernel: [ 1436.732604] fe40: 00000000 00000000 adb2fe74 adb2fe58 806fe570 806d2450 00000001 adb2fe84
Apr 15 17:34:04 octopi kernel: [ 1436.737137] fe60: adb2fef0 00000000 adb2fe84 adb2fe78 80657c84 806fe534 adb2fedc adb2fe88
Apr 15 17:34:04 octopi kernel: [ 1436.741810] fe80: 80657d2c 80657c6c 00000000 00000000 00000001 0001a934 0000a624 adb2fee8
Apr 15 17:34:04 octopi kernel: [ 1436.746644] fea0: 00000001 80b8cd40 00000000 00000000 00000000 adb2ff08 adafc9c0 adb2ff80
Apr 15 17:34:04 octopi kernel: [ 1436.751664] fec0: 00000000 00000000 00024f58 adb2e000 adb2ff4c adb2fee0 8028a930 80657ca0
Apr 15 17:34:04 octopi kernel: [ 1436.756743] fee0: 00024f58 b3129e00 729004f0 00024f58 00000001 00000000 00024f58 adb2fee8
Apr 15 17:34:04 octopi kernel: [ 1436.761925] ff00: 00000001 80b8cd40 adafc9c0 00000000 00000000 00000000 00000000 00000000
Apr 15 17:34:04 octopi kernel: [ 1436.767280] ff20: 00000000 00000000 adafc9c0 00024f58 adafc9c0 729004f0 adb2ff80 80108224
Apr 15 17:34:04 octopi kernel: [ 1436.772635] ff40: adb2ff7c adb2ff50 8028ab50 8028a85c 802aa234 802a9a10 adafc9c0 adafc9c1
Apr 15 17:34:04 octopi kernel: [ 1436.777992] ff60: 00024f58 729004f0 80108224 adb2e000 adb2ffa4 adb2ff80 8028ada0 8028aaa8
Apr 15 17:34:04 octopi kernel: [ 1436.780454] ff80: 00000000 00000000 00023120 00024f58 727fe0e8 00000004 00000000 adb2ffa8
Apr 15 17:34:04 octopi kernel: [ 1436.782770] ffa0: 80108080 8028ad58 00023120 00024f58 0000000b 729004f0 00024f58 00000000
Apr 15 17:34:04 octopi kernel: [ 1436.785067] ffc0: 00023120 00024f58 727fe0e8 00000004 727fe530 7611424c 00000138 00026be0
Apr 15 17:34:04 octopi kernel: [ 1436.787362] ffe0: 00000000 727fe0b8 00000000 76ea6314 80000010 0000000b 00000000 00000000
Apr 15 17:34:04 octopi kernel: [ 1436.789670] [<806ce384>] (tcp_push) from [<806d1950>] (tcp_sendmsg_locked+0x308/0xdfc)
Apr 15 17:34:04 octopi kernel: [ 1436.791968] [<806d1950>] (tcp_sendmsg_locked) from [<806d247c>] (tcp_sendmsg+0x38/0x4c)
Apr 15 17:34:04 octopi kernel: [ 1436.794261] [<806d247c>] (tcp_sendmsg) from [<806fe570>] (inet_sendmsg+0x48/0xe4)
Apr 15 17:34:04 octopi kernel: [ 1436.796554] [<806fe570>] (inet_sendmsg) from [<80657c84>] (sock_sendmsg+0x24/0x34)
Apr 15 17:34:04 octopi kernel: [ 1436.798842] [<80657c84>] (sock_sendmsg) from [<80657d2c>] (sock_write_iter+0x98/0xcc)
Apr 15 17:34:04 octopi kernel: [ 1436.801128] [<80657d2c>] (sock_write_iter) from [<8028a930>] (__vfs_write+0xe0/0x138)
Apr 15 17:34:04 octopi kernel: [ 1436.803411] [<8028a930>] (__vfs_write) from [<8028ab50>] (vfs_write+0xb4/0x1bc)
Apr 15 17:34:04 octopi kernel: [ 1436.805700] [<8028ab50>] (vfs_write) from [<8028ada0>] (SyS_write+0x54/0xb0)
Apr 15 17:34:04 octopi kernel: [ 1436.806880] [<8028ada0>] (SyS_write) from [<80108080>] (ret_fast_syscall+0x0/0x28)
Apr 15 17:34:04 octopi kernel: [ 1436.809116] Code: e155000c 03a05000 e2117902 1a000039 (e5d50028)
Apr 15 17:34:04 octopi kernel: [ 1436.810307] ---[ end trace 86badae211150a02 ]---```

jacksonliam, 2018-04-15:
Are you using a USB WiFi dongle? I believe this can be caused by high wifi throughput. 

Possibly fixed in newer kernels/OSs, are you running the latest octopi?

cassianoleal, 2018-04-15:
Everything is up to date. WiFi is the onboard module of the RPi3B+.

How can I monitor for that?

jacksonliam, 2018-04-16:
OK cool, not USB wifi then!

I think you have hit this bug, with the new Pi 3B+ 
https://github.com/raspberrypi/linux/issues/2481

An rpi-update might fix it. 

Either way, I don't think it's an mjpg-streamer issue, so I'll close this issue.

closed 2018-04-16 by jacksonliam