jacksonliam/mjpg-streamer #131: [REQUEST] Image Watermark and/or Date/time stamp and/or custom text watermark

bolsoncerrado, 2018-04-19:
Would it be possible to have the image stream from the camera attached to a Raspi with either or all of the following?

- Custom image (and custom positioning)
- Date/time (systime) stamp (and custom positioning, perhaps custom fonts too?)
- Custom fixed text (user fills in a small text field, custom font?)

Thanks!

dvv, 2018-04-19:
Any of these would mean to unpack, modify,  and pack jpegs coming from camera hardware.
This in fact mean that it is not just the streamer.
I would resolve your probs via ffmpeg.

jacksonliam, 2018-04-19:
This is true, might be possible to add something to the raspicam plugin that does it in the GPU but its not something I have the time or interest to add right now!

I'd recommend using an overlay on the display side.

closed 2018-04-19 by jacksonliam