jacksonliam/mjpg-streamer #140: The bottom one third of a v4l2 device frame is displayed as static

The-Oracle, 2018-06-08:
So I am trying to output to http using a uvc input (v4l2loopback device), I can get the stream up and running, but the bottom third of the frame is showing up as static, like you'd see on an old analog OTA TV set. When I view the V4L2 device in VLC, the stream shows correctly. Any ideas what is going on?

I did have to add RGB24 Pixel format support, as my V4L2 is a loopback device fed by a capture utility which obtains raw 14-bit pixel data from an infrared camera over SPI, and the capture utility colorizes the raw 14-bit thermal data with the thermal Fusion Palette. I'm just at a loss why the bottom third of the frame shows as static, but the top two thirds displays correctly.

Here is a sample photo of the mjpg-streamer output:

[image]

The only real substantial addition to code was to jpeg_utils.c, the rest was just adding support of V4L2_PIX_FMT_RGB24 as a second fourcc pixel format option.

`else if (vd->formatIn == V4L2_PIX_FMT_RGB24) {
        while(cinfo.next_scanline < 120) {
            unsigned char *ptr = line_buffer;

            for(int x = 0; x < vd->width; x++) {
                *(ptr++) = yuyv[0];
                *(ptr++) = yuyv[1];
                *(ptr++) = yuyv[2];
                yuyv += 3;
            }

            row_pointer[0] = line_buffer;
            jpeg_write_scanlines(&cinfo, row_pointer, 1);
        }
    }`

rhobincu, 2018-06-08:
You should use `while (cinfo.next_scanline < cinfo.image_height)`

Also, can't you use a memcpy instead of copying byte-by-byte?

`memcpy(line_buffer, yuyv, 3 * vd->width);`

The-Oracle, 2018-06-10:
I had hard coded in the height of the image, thinking maybe it wasn't getting passed the correct number of rows, thats why I had the 120 there instead of vd->height. 

I switched to using memcpy, but unfortunately it has the same result. 

I tried encoding the V4L2 frames as JPEGs in my capture utility, and it now displays perfectly. Thanks for the help!

closed 2018-06-10 by The-Oracle