jacksonliam/mjpg-streamer #142: Maximum output_file size of 2GB?

formidable01, 2018-06-11:
I am running mjpg-streamer with three  plugins active on my RPi. The command line is:

mjpg_streamer -i `input_raspicam.so -x 1280 -y 720 -ISO 25 -sh 75 -co 100 -br 70 -quality 10 -framerate 10 -cs ${this.cameraID}` -o `output_ws.so -p ${this.wsPort}` -o `output_file.so -f /home/stir/videos -m video.${this.cameraID}.${d}.mjpeg -o  `output_http.so -w /home/pi/www${this.cameraID} -p ${this.httpPort}`

I'm doing a lot of stuff streaming to a web socket, collecting images to a file and streaming to a HTTP port so I can easily capture stills. And it all works great. But I recently noticed that the MJPEG output_file collection stops at around 2GB. I looked at the code and can't see any obvious reason for that. It does align closely with the RAM size of the RPi. Am I exceeding the max_frame_size? Is there a way to work-around this?

formidable01, 2018-06-12:
This appears to be 32 bit Linux LFS issue. The make flow should be modified to use the following gcc compile switch -D_FILE_OFFSET_BITS=64 when building on RPi (for instance)

jacksonliam, 2018-06-12:
Ahh yes, great investigation, thanks!

I think this can be also fixed with a #define in the source file.

formidable01, 2018-06-12:
I found that in addition to the compile switch  -D_FILE_OFFSET_BITS=64, these also need to be defined -D_LARGEFILE_SOURCE and -D_LARGEFILE64_SOURCE  I tested the rebuilt plugin and it successfully generates file >2GB

jacksonliam, 2018-06-12:
Thanks, I'll leave this issue open until I get a chance to fix it in the source!

CarlosGS, 2018-07-21:
Hi! Where should we put the mentioned flags?

nybblecat, 2018-07-21:
Hi Carlos,

You should be able to use the compile switch flags mentioned in this thread
by using the advanced compilation instructions using cmake:

https://github.com/jacksonliam/mjpg-streamer#advanced-compilation-via-cmake

CarlosGS, 2018-07-21:
Ahh, thats it!! Thank you for the clarification :-)

paulhayes, 2020-02-27:
I've had the same issue, so thought I'd follow the advice given above, but without any success.

I tried running cmake with the following flags:
```
cmake -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE=1 -D_LARGEFILE64_SOURCE=1 ..
```

I've also tried adding the following to ```CMakeLists.txt``` 
```
add_definitions(-D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64)
```

Both methods successfully built, but when testing I still get the error while recording to disk ```write(): File too large``` 

Does anyone know what I'm doing wrong?