jacksonliam/mjpg-streamer #143: Tag new release

craftyguy, 2018-06-16:
Hi,
This application is currently an orphan in Arch Linux's AUR: https://aur.archlinux.org/packages/mjpg-streamer/

I would like to adopt it, but it would be much easier to maintain if there were a tagged release for it. Is there any chance you could do that?

ChaosEngine, 2020-05-23:
Same thing here for Gentoo Linux. It would be wonderful if we would have tagged release on which we could be basing our [ebuild](https://packages.gentoo.org/packages/media-video/mjpg-streamer) package.

jacksonliam, 2020-12-18:
Added a tag, sorry it took so long I wanted to make sure we had a stable thing to tag! I will try to remember to add new tags every nown again.

closed 2020-12-18 by jacksonliam