jacksonliam/mjpg-streamer #147: output to http has noticeable delay

BDHU, 2018-07-20:
I'm running a stop watch on Windows 7 and recording the timer with a camera, the resolution is set to 320x240. However, whenever a picture is sent back through local network to my testing machine it seems like there is always a 0.3 seconds delay which is quite noticeable. Is this a normal feature? I'm curious if anyone else has the same issue because I' trying to find out if my network setting creates the overhead. Thanks!

jacksonliam, 2018-07-20:
Yes some delay is normal, it's not designed for low latency.

0.3s sounds pretty good to me.

BDHU, 2018-07-31:
The latency issue seems like a big problem to me. I timed the execution time of `uvcGrab` in `v412uvc.c` and every single call to that function costs about 70ms, which contributes half of the latency to total delay. Is there any way to improve the performance or do you have any recommendations? Thank you.

jacksonliam, 2018-07-31:
That call is just waiting for a frame from your camera

If you are running at 15fps then ~70ms seems about right.

BDHU, 2018-08-03:
From the log it seems the frame rate is coerced back to 30 even though I set it as 15 when initializing the script. The problem was with `int ioctl(int fd, VIDIOC_DQBUF, struct v4l2_buffer *argp)` in uvcGrabframe() which takes the majority of the overhead. If the frame rate is forced back to 30 how does this make sense?