jacksonliam/mjpg-streamer #152: jpg pictures instead of mpg video

emetss, 2018-07-30:
I'm using ./mjpg_streamer -o "output_http.so -w ./www" -i "input_raspicam.so" with my picamera on octoprint. I get mpg video. I need jpg pictures instead. How to do that please?

nybblecat, 2018-07-30:
Hi there,

You should be able to go to [your URL]/?action=snapshot and get a JPG
snapshot of your cam.

Check out the README for more details:

https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/output_http/README.md#browservlc

emetss, 2018-07-30:
Thanks. I did. I tested the URL and I saw picture. However It doesn't save images at the end of the printing. It only saves mpg file. Maybe this is octoprint issue.
I'm using the following to run the camera:
/mjpg_streamer -i "./input_raspicam.so -fps 20 -x 1280 -y 720" -o "./output_http.so"
I tried also -usestills. I should get pictures under home/pi/.octoprint/timelapse/tmp but I got nothing.

jacksonliam, 2018-07-30:
Not an mjpg streamer issue. Octoprint is what saves the images to that folder.

closed 2018-07-30 by jacksonliam

Dreanaught, 2018-08-02:
@emetss you might also look at the output_file plugin.
Your call might look like this:
/mjpg_streamer -i "./input_raspicam.so -fps 20 -x 1280 -y 720" -o "./output_http.so" -o "output_file.so -f ./home/pi/.octoprint/timelapse/tmp -d 15000"