jacksonliam/mjpg-streamer #160: Make errors

djkraven, 2018-08-27:
I am trying to make this on a Linux Mint system get the following. The Warning is easily resolved with a space. The Errors I am not as sure on the resolution. 

Warning:
Building CXX object plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o
/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:86:5: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
     " Help for input plugin..: "INPUT_PLUGIN_NAME"\n" \
     ^
Errors:
Building CXX object plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o
/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp: In function ‘void* worker_thread(void*)’:
/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:408:34: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
     compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
                                  ^~~~~~~~~~~~~~~~~~~~~~~
/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:408:34: note: suggested alternative: ‘IN_CMD_JPEG_QUALITY’
     compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
                                  ^~~~~~~~~~~~~~~~~~~~~~~
                                  IN_CMD_JPEG_QUALITY
plugins/input_opencv/CMakeFiles/input_opencv.dir/build.make:62: recipe for target 'plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o' failed
make[2]: *** [plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o] Error 1
CMakeFiles/Makefile2:237: recipe for target 'plugins/input_opencv/CMakeFiles/input_opencv.dir/all' failed
make[1]: *** [plugins/input_opencv/CMakeFiles/input_opencv.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

jacksonliam, 2018-10-21:
Is this now fixed?

rmccartney856, 2018-10-21:
I've experienced the same error building on a Raspberry Pi running raspbian-strech. Here's my command line error from the build.

 `[ 31%] Building CXX object plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o
/home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:86:5: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]
     " Help for input plugin..: "INPUT_PLUGIN_NAME"\n" \
     ^
/home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp: In function ‘void* worker_thread(void*)’:
/home/pi/mjpg-streamer/mjpg-streamer-experimental/plugins/input_opencv/input_opencv.cpp:408:34: error: ‘CV_IMWRITE_JPEG_QUALITY’ was not declared in this scope
     compression_params.push_back(CV_IMWRITE_JPEG_QUALITY);
                                  ^~~~~~~~~~~~~~~~~~~~~~~
plugins/input_opencv/CMakeFiles/input_opencv.dir/build.make:62: recipe for target 'plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o' failed
make[3]: *** [plugins/input_opencv/CMakeFiles/input_opencv.dir/input_opencv.cpp.o] Error 1
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
CMakeFiles/Makefile2:237: recipe for target 'plugins/input_opencv/CMakeFiles/input_opencv.dir/all' failed
make[2]: *** [plugins/input_opencv/CMakeFiles/input_opencv.dir/all] Error 2
make[2]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:127: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
Makefile:17: recipe for target 'all' failed
make: *** [all] Error 2
`

minhnhattonthat, 2018-12-30:
IMO this error related to incompatible of openCV 4, as also mentioned [here](https://github.com/robotpy/robotpy-cscore/issues/39).
So the options are to downgrade to openCV 3, or build with opencv plugin disabled. The later can be done with the following script
```
cd mjpg-streamer-experimental
cmake -DPLUGIN_INPUT_OPENCV=OFF
make
sudo make install
```
Of course this is just for the case that you don't need to use the openCV plugin.

jacksonliam, 2020-07-03:
I think I have pushed a change to help with this, limiting the building of the opencv plugin to opencv 4. Could someone with the error pull the latest, do a `make distclean` and try again?