jacksonliam/mjpg-streamer #162: 403: Forbidden! frame already sent

djkraven, 2018-09-02:
What be the cause for getting this error when a browser one machine is streaming the cam without issue but other machines the error. Multiple reloads the other machines will eventually start streaming.

djkraven, 2018-09-20:
This seems to be a problem when viewing the cam over the internet from a NAT address on several different machines and the  check_client_status using IP as the unique client indicator. I originally had the cam running on a RaspberryPi without this issue. Now it is on a netbook with LinuxMint 19 32bit.

jacksonliam, 2018-10-19:
I'm not sure what the code is for, perhaps some kind of denial of service protection?

Try changing https://github.com/jacksonliam/mjpg-streamer/blob/f387bb44e6c087271b763b27da998bf2e06c4f5d/mjpg-streamer-experimental/plugins/output_http/httpd.c#L375
To `return 1` and re-building.

djkraven, 2018-10-24:
I have tested with a 64b machine and this seems to help avoid the error and keeping streams fluid. Need to do it on a 32b machine to fully confirm. Thank you for the pointer of what to change.

schiffsratte, 2020-06-27:
Works for me too when streaming through NAT

LMBernardo, 2022-09-04:
Does anyone have any idea why this would be desirable? It seems sort of pointless - and definitely worth a PR imo. Specifically, removing the various checks such as the one at https://github.com/jacksonliam/mjpg-streamer/blob/85f89a8c321e799fabb1693c5d133f3fb48ee748/mjpg-streamer-experimental/plugins/output_http/httpd.c#L1071

LMBernardo, 2022-09-04:
HTTP rate limiting (i.e., "403: Forbidden! frame already sent") is now disabled by default in [my fork](https://github.com/LMBernardo/mjpg-streamer), where it is now an optional command line parameter.